opendata-stuttgart/sensors-software #718: Limit on number of OLED display pages?

Bergerie, 2020-05-07:
My set-up consists of:
DS011
OLED SSD1306
DS18B20
BME280

The OLED currently limits to four displays, e.g. Device Info, SDS011, BMP/E280, WiFi Info. If I exclude the BMP/E280 device, the DS18B20 info will be displayed instead. However, putting the BME/E280 device back, the display takes precedence to BMP/E280 device. Is there a way the number of display will be increased such that the info associated with the Ds18B20 will equally pass through.

ricki-z, 2020-05-07:
Short answer: No.
We had a version that displayed all sensors. The first wish of many users was to reduce this so that a complete cycle doesn't need so long.

Bergerie, 2020-05-08:
This is a pitty. I put the OLED display in place for debugging purposes. Is there a way you can lead me to enabling in excess of 4 displays? Based on installed components I need 5. Appreciate any assistance you can give.

dirkmueller, 2020-09-02:
does it have to be on the display ? you can see all the data including firmware status data and error states from the webui.

Bergerie, 2020-09-03:
What’s the point of having an oled display. If I’m in the field a web interface is not handy, a simple display surely is. So in short, YES I at least have a need for it.

dirkmueller, 2020-09-03:
Okay, thanks for clarifying. I'll see what can be done about it, but the display code is pretty messy already :/

Bergerie, 2020-09-05:
Sounds promising, look forward to whatever solution. Agree with the “ code is pretty messy already” though the core for implementation appears to be there. I did not want to temper with the code since this would be overwritten anyway during any future OTA update.