opendata-stuttgart/sensors-software #721: IQAir integration

kaimi-, 2020-05-16:
Hello,
Is it possible to integrate sensor with IQAir (AirVisual) service?
By possibility I mean whether the client device (AirVisual Pro) was researched in terms of network requests (HTTP, whatever) used for data transmission? 
If such research will be done, there will be no problems in legal aspect with such integration?

I understand, that product identifier (unique key) is required to register sensor within IQAir service.

Thank you.

dirkmueller, 2020-09-15:
@kaimi- this is a community effort, please don't make use search documentation .. help us find the right pointers. from what I can find so far, airvisual provides an api to query its own database, but not to submit. submission seems to be bound to their set of devices. 

if I'm wrong on this, please help us find the right api documentation and requirements and legal requirements for this integration.

KoffeinKaio, 2020-09-16:
It would be the easiest way to just contact their support and ask if its possible.
If there even is an API available/planned to send 3rd party data to.

Doubt they will be happy if we reverse their API/HTTP(s) communication.