opendata-stuttgart/sensors-software #725: open access point configuration page has WiFi password embedded

carstenjahn, 2020-05-24:
A few reported issues for this project discuss that the WiFi password will be transmitted unencrypted during initial setup, because neither WPA2 nor HTTPS are used.

In my view, it is a much bigger issue that the firmware goes into "open access point" mode if it cannot reach the previously configured network. (This may happen because it is scheduled to be offline at night etc.) This access point can be reached by everybody, even without packet sniffing. Its configuration page reveals which WiFi network SSID is configured, and the page source code contains the WiFi password (you can see it if you use the browser developer feature "Inspect Element" on the password field, or save the page as HTML and view it there, look for the "wlanpwd" input field).

The password should be transmitted from server to browser as "***". I think this has been discussed for the basic auth password field, but it should be implemented for the WiFi password field as well.

I'll try the basic auth mechanism to overcome this security issue (the page says "only if you really know what you're doing" - but it should rather encourage people to set it!) and also use a separate guest WiFi password for this device.

saper-2, 2020-06-07:
Hi,
I am about to build my own sensor, @carstenjahn  thank you for pointing this out, I'll have to not forget to configure security in my sensor 😄 .
Pointing out a problem is good 👍 , but without a idea of solution it's harder on programmer(s) to fix it 😄 , so let me add my "idea", I setup a lot of esp8266 with Tasmota firmware and they have implemented very good solution for this problem:
If the wifi network is not connectable the esp tries looking/connecting to that specified network. If the device running tasmota is powered off 5 times in row it switch back to AP mode, and allow to connect to it's AP and reconfigure it.

So here is my idea based on Tasmota solution:
The sensor will try find & connect to saved wifi network indefinitely.
If it fail to connect, and it will be powered on 5 times without being able to connect to remembered network it should switch then to AP mode with "blank" wifi password. So the "reboot counter" is reset on successful wifi connection (connected to wifi and received IP from DHCP server).
/or/
The sensor will try find & connect to saved wifi network indefinitely.
If I press & hold button "Flash" (NodeMCU board) for e.g. 20sec - the senor will switch to AP mode.

dirkmueller, 2020-08-31:
I believe I already fixed this issue in the beta firmware:

```
NRZ-2020-130-B4
* Prevent password leakage in configuration dialogs
```

dirkmueller, 2020-09-02:
Please reopen if there is still an issue with beta firmware.

closed 2020-09-02 by dirkmueller