opendata-stuttgart/sensors-software #726: InfluxDB mean value of day and night

atze09, 2020-05-25:
Hello guys,
I successfully implemented my DNMS Sensor with InfluxDB on home assistant. 
Now i have the questions how to handle the InfluxDB requests.

As for the noise sensor, it would be interesting to have the mean value of Laeq for daytime and nighttime. So from 6to 10 o clock and from 10 to 6.
Does anyone has already such a query, or can help me get this fixed?
This is my current InfluxDB:
`SELECT mean("DNMS_noise_LA_max") AS "mean_DNMS_noise_LA_max", mean("DNMS_noise_LA_min") AS "mean_DNMS_noise_LA_min", mean("DNMS_noise_LAeq") AS "mean_DNMS_noise_LAeq" FROM "luftdaten"."autogen"."luftdaten" WHERE time > :dashboardTime: AND time < :upperDashboardTime: GROUP BY time(:interval:) FILL(null)`

Thank you in advance

hbitter, 2020-05-31:
Hello atze09,
have a look at [this page](https://www.rexfue.de/). Reinhard Fürst realized a special noise reporting page [link](https://laerm.citysensor.de/) where the hourly values, daily values, day- night-values and the LDEN indes is schown based on the LAeq vales from the archive of all registered DNMS noise sensors.
Hope this helps,
Helmut

atze09, 2020-05-31:
Hello Helmut,
Thank you for this awesome hint!
Best regards,
Dominik

rexfue, 2020-05-31:
Hi, please note: Since my servers are working on the edge of exhaustion (!), the daily averages are not (always) correct. But the values of the previous day(s) are always OK, because I update the values of the previous day every day around 11am. 
@hbitter Thanks for mentioning my site.

atze09, 2020-06-04:
> Hi, please note: Since my servers are working on the edge of exhaustion (!), the daily averages are not (always) correct. But the values of the previous day(s) are always OK, because I update the values of the previous day every day around 11am.
> @hbitter Thanks for mentioning my site.

Awesome work Reinhard!
Is it possible that you can assist me via private message how to get the data out of my influx db?
i am a database newbie and would like to have the data analysis local here on my server.
Thank you in advance.
Best regards,
Dominik

rexfue, 2020-06-04:
@atze09 OK, let's try it. I don't know influxDB but I know the formulas. email: rexfue@gmail.com

ricki-z, 2020-09-14:
@atze09 can we close this issue?

closed 2020-09-14 by atze09