opendata-stuttgart/sensors-software #728: Support of Ethernet (Olimex ESP32 POE ISO)

tilman1, 2020-06-04:
Hi 
I would like to run the sensor firmware on Olimex ESP32 POE ISO via ethernet rather than WiFi.
I have cloned beta branch and added the ethernet support there.  I would like to put it back into this repository as new branch, so that I can test it more  for side effects before it can be merged onto the beta branch.
I have little to none experience with github. I seem not to have permission to add a branch -- how can I add a branch?

Thanks
TIlman

nagyrobi, 2020-06-27:
Please also add support for [wESP32](https://wesp32.com/). Same as Olimex, only difference is that the Ethernet is connected to pins 16 for mdc and 17 for mdio.

DeeKey, 2020-07-20:
Use pull requests

bertrik, 2020-07-20:
> I have little to none experience with github. I seem not to have permission to add a branch -- how can I add a branch?

In short:
* fork the opendata-stuttgart archive into your own account, on github
* clone it locally on your machine: git clone <your own github archive>
* checkout the beta branch: git checkout beta
* create a branch from beta: git checkout -b <name_of_your_feature>
* commit your changes with a clear description, e.g.: git commit .
* push it to your own archive : git push (and follow the instructions to push to a branch on your github account)
* on your github account you'll see a message that you just pushed to a new branch, follow that to create a merge request towards the opendata-stuttgart project

The above is not meant as a complete guide, just a summary of the steps.

nagyrobi, 2020-07-20:
Easy, yeah.

tilman1, 2020-07-22:
Hi guys,
I did all steps suggested by bertrik but the last step. I would like to do some more testing before issuing a pull request. If somebody would want to test -- the  branch is called OlimexEthernet. 
@bertrik 
Thanks for the hints. That helped

@nagyrobi 
Fpr supporting the wESP32 would need to capable of distinguishing  between the different hardwares. Any idea how the program can find out whether it runs on an olimex or an wESP32?

Thanks and best regards
Tilman

nagyrobi, 2020-07-24:
I think this should be done manually. Put a checkbox in settings: "Ethernet support: ( ) Olimex (x) wEESP32". This would allow further addition of supported hardware later.

tilman1, 2020-08-10:
I have implemented the configuration via   ethernet  in analogy to that of the WLAN, i.e. if the device is not configured, both, the ethernet and the WLAN become active. In the moment one network type gets connected, the other one is deactivated, and the device can be configured using the WEB GUI. If one does not want to give up on that concept, the fw needs to know on which device it is running.

tilman1, 2020-11-08:
@nagyrobi
 mdc and mdio are part of the Media Independent Interface Management (MIIM).
 https://en.wikipedia.org/wiki/Media-independent_interface

Maybe, there is not really a problem.
For configuring the ethernet, I use the standard infrastructure provided via the arduino IDE, i.e. the configuration code of the ESP32 pins to communicate with a LAN7810 is not really visible or accessible on the application layer. Apparently this is encapsulated by the board support packages:  (see also https://wesp32.com/software/#warning-gpio-pin-configuration). Can you try to get one of the ethernet examples delivered with the arduino IDE to run?  Once you have configured your board , you should find  it in File->Examples->Wifi->ETH_LAN8720