opendata-stuttgart/sensors-software #729: Lokaler NTP Server nicht nutzbar NRZ-2020-130-B6/DE Local NTP server cannot be used

user915438, 2020-06-05:
Hallo,
ich nutzer einen lokalen NTP-Server, den ich gern eintragen möchte. Ist es möglich eine Abfrage unter "Konfiguration" einzufügen.

Vielen Dank.

Hi,
I am using a local NTP server that I would like to register. Is it possible to insert a query under "Configuration".

Thanks a lot.

ricki-z, 2020-06-08:
Hello,
we have only limited space available in the Flash and RAM. So such an option for a very limited amount of users (you are the first asking for this) wouldn't have a very high priority.
We are using NTP servers from ntppool.org, and if they aren't available the local gateway.

nagyrobi, 2020-06-27:
If you router supports it, you can redirect all requests to UDP port 123 to your preferred NTP server. Your IoT devices won't even notice that somebody else answered.

dirkmueller, 2020-09-02:
@user915438 ist der lokale NTP server das default gateway oder ein andere IP im lokalen netz? ein fallback für default gateway (da viele geräte das verwenden) sollte möglich sein als config option. 

regarding @ricki-z getting the system time is really important for OTA to work (requires https with validated certs which means system time needs to be right).

sumerland, 2020-09-04:
In my setup (Asus router with ntp traffic redirected to local ntp server) the sensor fails to get ntp updates as soon as the redirect is enabled. This leads to frequent restarts (every 2...3 hours). I need to switch redirect off - now the sensor has been running for a couple of days (until today's OTA).

dirkmueller, 2020-09-04:
Can you please test again with the new beta and the orevt setting if the instability reappears? I would like to know more then, like serial output or at least a status page screenshot

sumerland, 2020-09-04:
Looking good now! After a restart the sensor successfully connects to 192.168.1.1. Many thanks!

dirkmueller, 2020-09-15:
thats part of #794 . thanks for the feedback! closing this issue for now.

closed 2020-09-15 by dirkmueller