opendata-stuttgart/sensors-software #737: Keine PM Werte mehr

ammerscm, 2020-06-13:
Seit der neuen Firmware NRZ-2020-130-B6/DE gibt die Messstation keine Feinstaubwerte mehr aus.
Die Luftfeuchtigkeit von 99% scheint auch nicht zu stimmen.

ricki-z, 2020-06-14:
Welcher Sensortyp wird verwendet (SDS011,SDS30,PMS?003)? Ist dieser in der lokalen Konfiguration auf dem Sensor aktiviert (kann nach einem Update bei den Beta-Versionen passieren, daß er da was "vergisst").
Bei der Luftfeuchte kann es beim DHT22 nach einiger Zeit zu einer Sättigung kommen und er zeigt dann nur noch 99% an. Hier wäre ein Wechsel auf den BME280 zu empfehlen.

ammerscm, 2020-06-14:
SDS011 und DHT22.
Ist alles korrekt in der Konfig eingetragen.

ricki-z, 2020-06-15:
Die Kabel wurden auch schon geprüft?
Wir haben nur ein paar Leute, die die Beta Version nutzen. Aber ich habe von keinem gehört, daß es Probleme mit dem SDS011 gibt (zumal gerade dieser ja unser "Hauptsensor" ist und dessen Funktion immer getestet wird).

ricki-z, 2020-06-15:
Werden auch lokal, über die Seite "Aktuelle Werte", keine Werte ausgegeben?
Aktuell ist der Empfang von Daten auf Madavi.de deaktiviert, da der Server ein Lastproblem hat. Daher sind dort die Grafiken derzeit leer.

Bratkartoffl, 2020-07-18:
Ich habe dasselbe Problem:

**Lokale Werte** 
SDS011 | PM2.5
-- | --
SDS011 | PM10 | - µg/m³
-- | -- | --

DHT22 | rel. Luftfeuchte | **99.9 %**
-- | -- | --

Der Sensor hat von einem Tag auf den anderen aufgehört zu funken. Habt ihr eine Lösung gefunden?

ricki-z, 2020-07-19:
Wurden die Kabel am SDS011 kontrolliert? Die lösen sich manchmal durch Vibrationen (Lüfter) oder Erschütterungen.

Bratkartoffl, 2020-07-19:
Nein noch nicht, da ich dazu das Gehäuse öffnen müsste. Gibt es eine Möglichkeit in die Logs oä zu schauen?

imaluschi, 2020-08-24:
Selbiges siehe #759, habs leider zu spät gesehen...

dirkmueller, 2020-09-02:
dies war eine regression von https://github.com/opendata-stuttgart/sensors-software/pull/769 . es wurden global parameter geändert die dazu geführt haben dass die kommunikation mit dem SDS nur noch mit 115200 baud läuft. was mit der alten arduinocore version nicht geht. 

das problem ist bereits gefixed.

closed 2020-09-02 by dirkmueller