opendata-stuttgart/sensors-software #738: SPS30

unixweb, 2020-06-14:
Habe einen SPS30 Feinstaubsensor hier und einen NodeMCU V2. 
Angeschlossen ist er wie hier empfohlen : 
Pinout: 1 2 3 4 5

    Pin 1 (5V) -> Pin VU/VIN
    Pin 2 (SDA) -> Pin D3 (GPIO0)
    Pin 3 (SCL) -> Pin D4 (GPIO2)
    Pin 4 (SEL) -> Pin GND
    Pin 5 (GND) -> Pin GND

Hab probeweise ESP-Easy installiert um zu sehen ob das Gerät erkannt, wie im Bildschirmfoto zu sehen,  wird das Gerät unter 0x69 erkannt.
[image]
Schließe ich den Sensor mit der Firmware von Airrohr an wird der Sensor nicht erkannt. 
Was ist nun falsch oder was muss ich noch machen damit der Sensor erkannt wird.

Anbei noch ein paar Bilder des Sensors.

[image]
[image]
[image]

Grüße Joachim

jrichert67, 2020-06-20:
> Habe einen SPS30 Feinstaubsensor hier und einen NodeMCU V2.
> Angeschlossen ist er wie hier empfohlen :
> Pinout: 1 2 3 4 5
>
> ```
> Pin 1 (5V) -> Pin VU/VIN
> Pin 2 (SDA) -> Pin D3 (GPIO0)
> Pin 3 (SCL) -> Pin [...]

Hallo Joachim
Ich glaube der SPS30 ist etwas zickig. Ich habe auch einige Tage gebastelt und herausgefunden, dass er bei mir nur mit der letzten BETA FW sauber erkannt wurde. Ausserdem benötigte ich 10K Pullup Widerstände.
[image]
Im Moment habe ich auch ein weiteres Problem mit den Sensoren. Habe zwei Stück in absolut identischer Konfiguration. Der eine der beiden wahnsinnig hohe Werte. Hänge ich ihn an das Developer Kabel und lese die Daten mit der Sensirion SW aus, stimmt alles.
[image]

Viel Erfolg
Jochen

unixweb, 2020-06-21:
Hallo Jochen

Danke für Deinen Beitrag und Deine Mühe. 
Hab jetzt auch die BETA-Version installiert und siehe da, der Sensor wird erkannt.
Das mit den Kabel / Anschluß kann ich nicht erkennen. Bei mir funktioniert der Sensor immer 
auch mit dem Arduino Beispiel Sketch, es gibt keine Probleme. 

[image]
[image]

jrichert67, 2020-06-22:
Hallo Joachim
Gerne :-)
Vielleicht braucht es die Pullups auch nicht. Habe mir damals den Wolf gesucht und eventuell erst die Pullups und dann den Beta FW Update gemacht. Auf jeden Fall steht es so im Datenblatt und schaden kann es sicher nicht ;-)
Und betreibt das Teil ja mit 5 Volt. HAbe bei meinem  zweiten Sensor Judihui einfach mal alle Kabel SDA,SCL,GND und VCC durchgelötet und nicht gemaerkt, dass das 3 Volt waren. Danach hat der Sensor Wanhwitzige Werte im 10er Bereich geliefert. Nachdem mir das dann gedämmert war schnell auf 5V umgelötet und siehe da, läuft :-)
lG aus der Schweiz und viel Spass mit unserem gemeinsamen Hobby
Jochen

unixweb, 2020-06-22:
Hallo Jochen 

vielen Dank für Deine Nachricht und Erkenntnisse. 
Dass mit den 3 Volt war bei mir kein Problem da ich wusste dass der Sensor nur mit 5 Volt arbeitet. Am Pin VU der ESP8266 konnte ich die 5 Volt abgreifen. Die Pullups Widerstände hab ich ebenfalls seit Anfang an installiert. Erst als ich die BETA-FW installiert habe wurde der Sensor erkannt. 

Viel Spaß auch Dir weiterhin mit unserem gemeinsamen Hobby. 
Beste Grüße Joachim