opendata-stuttgart/sensors-software #739: NRZ-2020-130-B6/DE: BME 280 sendet zeitweise wirre Werte

dora71, 2020-06-17:
Hallo zusammen,

es gab schon 2 Issues, wo der BME 280 keine Daten mehr sendet. Meiner sendet zwar kontinuierlich, ab und zu sendet er jedoch wirre Werte. Im Anhang ein Screenshot.

Manchmal werden die Werte nach einiger Zeit wieder normal, manchmal hiflt nur ein Neustart des Sensors und danach läuft wieder alles.

[image]

nagyrobi, 2020-06-27:
Ihr Sensor muss defekt sein. Ich habe den gleichen Sensor auf der gleichen Firmware-Version wie Sie und es funktioniert perfekt.

dirkmueller, 2020-09-03:
It looks like we're reading an error code as temperature value. I can fix the readout to skip such values, so no measurements will be taken instead.

Given that it shows 100% humidity it is likely that the breakout board or the sensor is wet and electrically no longer isolated. I had this issue as well before reading the board with a waterproofing electronic spray that adds a plastic coating. You can use any kind of paint as well. Make sure to keep the breathing hole of the sensor itself untreated.

nagyrobi, 2020-09-03:
Can you please share with us the type/name of that spray?

dirkmueller, 2020-09-10:
I use this one: https://www.conrad.de/de/p/kontakt-chemie-plastik-70-74313-aa-isolier-und-schutzlack-400-ml-814169.html

But there's probably a million variants of these. It is highly electrically isolating and water proofing transparent spray.

You don't have to buy these Although I didn't try a normal fingernails finish would likely work as well. If the sensor isn't new you can see where it starts to oxidize and those areas need to be cleaned and treated

imaluschi, 2020-10-21:
It seems there is really a problem with BME280 values.
Had a look on another Sensor i changed the last time - added DNMS to SDS+BME. I didn't realize any corrosion. I used a hot glue pistol to protect the connection (and a little bit the resistors).
Auto Update was checked, it should be Firmware NRZ-2020-131.
[image]
[image]
Same with my own Sensor
[image]

Screenshots from 18th of october, already NRZ-2020-131.

[image]
[image]

Screenshots from 15th of september NRZ-2020-130B9.
[image]
[image]

Also with my former ID (had to change the ESP because during modification DNMS and BME280 in addition i destroyed the microUSB :-(  ). It was NRZ-2020-129.
[image]

The strange thing in my case:
Could it be that it has something to do with DNMS in addition?? When i have a look to the data of my colleagues sensor I do not realice such failure "measurements"/data previos to addingt the DNMS.

On the other hand, i found one such data of the sensor of another friend. In this case there are also a lot of WIFI-problems (Signal too bad, I already suggested a WIFI-Repeater).
Should have been Firmware NRZ-2020-129 (Auto-Update is configured).
[image]

And in the data of another sensor (which is not used at the moment), I also found some crappy BME280-data, also NRZ-2020-129.
[image]
[image]
In my opinion the sensor (data-failure shown in the last pictures) looks good - at least concerning corrosion.
[image]

Franz4596, 2020-10-21:
I have several BME280 in operation (e.g. ID: -1492670, #7067; -2281326, #3142 outdoor of my house), because DHT22 RHumidity drift to 100% after ~5month -BME drifts slower. 
The BME280's usually works well, but with some pieces I got wrong values -150°C / 600hPa too. Reboot or power off/on helps sometimes, about ~10% BME devices needs to replaced with new one.
1.####**ID: esp8266-1492670, #7067:** SDS011, DHT22,BME280,DNMS:
24.09.2017 BME280 dazu, weil DHT22-Feuchte sehr hoch
16.11.17 BME280 angemeldet bei luftdaten.org=>7067
06.06.18 BME280 Luftdruck ausgefallen (1,5bar)              16.07.18 BME280 ersetzt (rote Platine) 
15.01.19 BME280 t, p ausgefallen (-148°C, 800hPa)  23.02.19 BME280 ersetzt (rote Platine 3.3V) 
10.10.19 BME280 t, p ausgefallen (-148°C, 800hPa)  05.11.19 BME280 ersetzt (rote Platine 3.3V) 
07.09.20 1:00-17:00, 13.09.20 5:00... BME280 ausgefallen (Neustart)
14.09.20 BME280 ersetzt (rote Platine 5V) 

2.####**ID: esp8266-2281326** SDS011, DHT22,BME280, OLED1306:
05.12.2017:BME280 dazu
01.07.2018 BME-rot mit Kabel von {04} übernommen
02.10.-13.10.2019 keine Werte bme280 wegen Wackelkontakt ? Display war dunkel, DHT-RH=99%, reboot

I do not find the real fault reason, I guess its connection-corrosion or /and hardware. The error occurs before I add the DNMS one So, I solve problem with reboot first, if not help I replace BME280.
franz

Franz4596, 2020-10-22:
Maybe there is a power(5V/3.3V) problem?
My bme280 esp8266-1492670 shows error: at 7.9.10 ->reset ->9.9.10 ->reset ->14.9.10 ->changed DHT22!! and now BME280 works:
 [image]
current data/configuration:
[image]
Perhaps you change DHT22 ?

dirkmueller, 2020-10-22:
> A software-reboot did not help, it had to be a hard-reset (power off, power on). Now all Sensors work as expected again.

So thats useful to know. We do issue a soft-reset command to the BMX sensors on boot of the firmware, but we don't know whether it is processed (if it is processed, the sensor doesn't answer as it is booting, and if it doesn't, it might also not respond..). 

So this points at OTA being a trigger for the sensor to misbehave. I think we have no means currently to software-based ability to turn the power off for the sensor and recover it that way, as it is connected to permanently powered pins.

imaluschi, 2020-11-03:
Again BME as well as DNMS stopped working, UI was not very responsive - most of the time no response, only sometimes (maybe around measurements).

Screenshots from 03.11.2020:
[image]
[image]
[image]

Device status when i realized the occurence.
[image]

[image]

Tried software reset with no success:
[image]
[image]
[image]

Hard reset was necessary:
[image]
[image]

Now working again as expected again.

Overview: Soft-reset marked (around 17 o'clock), hard reset around 18 o'clock.
[image]

imaluschi, 2020-11-05:
And again. Same pictures
[image]
[image]
Hard reset necessary.

imaluschi, 2020-11-06:
Grrr, not very long since last hardware reset.
Maybe I will change the DHT-Sensor as suggested by @Franz4596 somewhen the next days. 

[image]
[image]

Franz4596, 2020-11-07:
Hi imaluschi, at 29.10.20 I looked at an esp8266-11732530 with SDS011,DHT22,BME280, and I found a bad bme280-value of 720hPa since 27.11.20 !?.   At 31.10.20 I dismount the DHT22 (because 100%RH), changed the BME280 with new one, insert a SHT31-DIS-F and a DNMS-T4.0. 
Maybe there is a comination of hardeware fault and software (timing)? 
In future I will never use DHT22(->100%RH), but SHT3x-Filter(t,RH) and try BME280 (pressure), SDS011 or SPS30. 
Please write a comment, when your DHT22-change or BME280-change work or not. 
Chart shows the bad values: ...13.2.20 bad DHT,  27.-31.10.10 bad BME280:
[image]

imaluschi, 2020-11-08:
Once again.
[image]
[image]

But now (2020-11-08 Data starting with 08:51 MEZ) changed the DHT-Sensor as @Franz4596 suggested.
Let's see if it will change. At the moment I have no other BME280 here (ordered, but maybe still some weeks until they arrive).

This time there was an "Exception" as reset reason. Had this already before, but also disfunction of BME and DNMS without Exception as reset reason.

imaluschi, 2020-11-08:
@dirkmueller , maybe an Issue with I2C-Bus? Both DNMS and BME280 are affected, and both are driven with I2C. SDS and DHT are working well.
Any suggestion what to do to find out more? I have to take look, maybe i have an BMP280 somewhere around.

Franz4596, 2020-11-08:
@imaluschi,  I have some new BME280 and SHT30 available. If you like I may send you one from 45968 Gladbeck?

imaluschi, 2020-11-09:
No, it was definitively not a broken DHT-Sensor. Thanks @Franz4596 for your offer, but as written i have an BMP280 somwhere around. But I think more and more it has something to do with I2C. When the BME280 stops to work as expected, the DNMS stops working totally. The only thing this two sensors have in common is use of the I2C-Bus.
[image]
[image]
One more hard-reset.

imaluschi, 2020-12-01:
The sensor worked nice for quite some days. About 19hours ago Firmware was auto-updated to NRZ-2020-133/DE (Nov 29 2020). But now same malfunction again - also with the new firmware.
As earlier, only hard reset solves the problem. Webpage is very unresponsive, the LED on the ESP is shining the whole time - left  the cover open. So if I see the LED shining the whole time, I know the problem occured again.
[image]
[image]

imaluschi, 2020-12-02:
Mine is not the only one.
Sensor esp8266-15570811, configuration: DNMS, SDS011, BME280

[image]
But in this case, it seems the Sensor is not "crashing" for quite some time and delivering at least sometimes (more or less) correct values. Even DNMS worked during this time, until 30th of November.
I got aware of this, because the DNMS (where I had a look on the values) is not working anymore.
[image]

imaluschi, 2020-12-09:
Another sensor with the same problem. The BME280 is NEW, so only some days placed outside.
Data from 09.12.2020
[image]
Same with other Sensors with configuration BME280 and DNMS, DNMS is failing. Strangely here have been some measurements in between.
[image]

pe0ter, 2021-01-18:
In a network of 20 identical airRohr stations using SDS011 and BME280, all installed at different addresses in a village and all connected to different wifi networks, 2 stations had their BME280 measurements go weird ( -147degC, 100%, 618hPa) on 12/01/2020 between 06:45 and 07:00.  I have asked the owners to power down/up their airrohr units and hope they will come back up normal. The only reason I can imagine this to happen is via an automatic firmware update of the airrohr units.

pjgueno, 2021-01-18:
You can also check if there is some oxydation on the solder points of the pins. A few month ago I had to wash the pins with  some electrocontact liquid in order to make a BME280 work again. After having washed the solder points you can put a small piece of duct tape on them to prevent them to oxydize again.

pe0ter, 2021-01-18:
Thank you @pjgueno 
Good tip, I will use nail varnish on my soder joints in future and if the power cycle does not solve the BME280 communication I will replace them and 'repair' the old ones by cleaning and varnish them too