opendata-stuttgart/sensors-software #740: Feinstaubsensor sendet keine Daten an sensor.community und Madavi

mcponline, 2020-06-18:
Hallo,
ich habe mir den Feinstaubsensor von luftdaten.info nachgebaut und wie in der Anleitung konfiguriert. Der Sensor an sich funktioniert, denn ich kann die Messwerte sehen wenn ich den Sensor direkt über sein IP aufrufe oder im Arduino-Monitor.
Aber auf Madavi und bei Sensor.community kann ich den Sensor mit seiner ID nicht finden. Die jeweiligen API's habe ich in der Konfiguration angewählt (beide mit HTTPS), ohne HTTPS funktioniert es auch nicht.

Der serielle Monitor aus der Arduino IDE gibt dabei folgendes aus:
13:13:21.433 -> Start reading SDS011
13:13:21.433 -> PM10 (sec.) : 1.30
13:13:21.466 -> PM2.5 (sec.): 1.00
13:13:21.466 -> PM10:  1.45
13:13:21.500 -> PM2.5: 1.00
13:13:21.500 -> ----
13:13:21.500 -> End reading SDS011
13:13:21.533 -> ## Sending to sensor.community - SDS011
13:13:21.667 -> Request failed with error: 403
13:13:21.701 -> Details:{"detail":"Node not found in database."}
13:13:21.736 -> Start reading DHT22
13:13:21.943 -> Temperature (°C): 22.10
13:13:21.943 -> Humidity (%): 64.40
13:13:21.977 -> ----
13:13:21.977 -> End reading DHT22
13:13:22.010 -> ## Sending to sensor.community - DHT22
13:13:22.147 -> Request failed with error: 403
13:13:22.183 -> Details:{"detail":"Node not found in database."}
13:13:22.215 -> ## Sending to madavi.de: 
13:13:40.775 -> Time for Sending (ms): 8775

Kann mir da jemand behilflich sein den Fehler zu finden?
Danke!

ricki-z, 2020-07-01:
Die Madavi API war ein paar Tage nicht erreichbar, weil der Hoster ein massives Problem mit seinem Speicher-Netzwerk hatte. Inzwischen sollte das wieder funktionieren.
Für die Sensor.Community API (ehemals Luftdaten.info) muss der Sensor auf https://devices.sensor.community/ angemeldet sein. Ohne Anmeldung erscheint die oben zu findende Meldung "Node not found in database."

mcponline, 2020-07-02:
@ricki-z 
Danke für die Info

closed 2020-07-02 by mcponline