opendata-stuttgart/sensors-software #742: NodeMCU board HW-628 doesn't work with NRZ-2020-130-B6

nagyrobi, 2020-06-27:
Using NodeMCU board HW-628. 

With NRZ-2020-129 it works.
With NRZ-2020-130-B6 it doesn't work.

Symptom: after flashing NRZ-2020-130-B6, board switches to AP mode as in the documentation. User interface at 192.168.4.1 shows up fine, I can select my WiFI network and add my network password, but after reboot, the board is inacceessible. It connects to the network because I see the DHCP has been requested but the board seems to be in a bootloop, because led flashes once in a while and if I ping it continuously there's rarely a moment when it replies to the ping.
But I am unable to open the web interface at all.

[image]

nagyrobi, 2020-06-27:
Serial output:
```
22:40:54.422 -> `⸮⸮⸮⸮FS...
22:40:54.422 -> 
22:41:12.273 -> failed to open config file.
22:41:12.273 -> output debug text to displays...
22:41:12.406 -> Connecting to Freifunk-disabled
22:41:12.869 -> ........................................
22:41:32.387 -> output debug text to displays...
22:41:32.421 -> Starting WiFiManager
22:41:32.421 -> AP ID: airRohr-XXXXXXX
22:41:32.454 -> Password: 
22:41:32.487 -> scan for wifi networks...
22:41:34.575 -> AP Password is: 
22:41:34.575 -> Starting Webserver... (IP unset)
22:41:50.453 -> ws: config page ...
22:41:50.453 -> begin webserver_config_body_get ...
22:41:51.978 -> ws: config page ...
22:41:51.978 -> begin webserver_config_body_get ...
22:41:53.801 -> ws: config page ...
22:41:53.801 -> begin webserver_config_body_get ...
22:41:53.934 -> ws: not found ...
22:41:53.934 -> ws: config page ...
22:41:53.967 -> begin webserver_config_body_get ...
22:41:54.862 -> wifi networks found: 4
22:41:54.862 -> ws: wifi ...
22:41:54.895 -> ws: config page ...
22:41:54.895 -> begin webserver_config_body_get ...
22:41:58.044 -> ws: not found ...
22:41:58.044 -> ws: config page ...
22:41:58.077 -> begin webserver_config_body_get ...
22:41:59.104 -> ws: config page ...
22:41:59.104 -> begin webserver_config_body_get ...
22:42:34.868 -> ws: config page ...
22:42:34.868 -> output debug text to displays...
22:42:34.902 -> Saving config...
22:42:35.067 -> Config written successfully.
22:42:35.067 -> output debug text to displays...
22:42:35.167 -> Restart.
22:42:35.764 -> H!⸮⸮x⸮⸮⸮⸮⸮airRohr: NRZ-2020-130-B6/EN
22:42:35.863 -> mounting FS...
22:42:35.896 -> opened config file...
22:42:35.929 -> parsed json...
22:42:35.929 -> output debug text to displays...
22:42:36.029 -> Connecting to ZACUT
22:42:36.525 -> ...........
22:42:42.093 -> WiFi connected, IP is: 192.168.1.102
22:42:42.127 -> Starting Webserver... 192.168.1.102
22:42:42.160 -> 
22:42:42.160 -> ChipId: XXXXXXX
22:42:42.195 -> 
22:42:42.195 -> MAC Id: ecfabcb3af26
22:42:42.195 -> Read DS18B20...
22:42:42.235 -> Send to :
22:42:42.235 -> sensor.community
22:42:42.259 -> Madavi.de
22:42:42.259 -> ----
22:42:42.259 -> SNTP synced: Sat Jun 27 20:42:42 2020
22:42:42.293 -> 

```
And it stops here.

ricki-z, 2020-07-01:
These boards seem to have a design failure. If you search for "HW-628" you will find many people with wifi problems.
I.e. http://forum.espruino.com/conversations/323241/

nagyrobi, 2020-07-03:
That's a superficial assumption, given that NRZ-2020-129 runs perfectly on it, and also [Tasmota](https://github.com/arendst/Tasmota) firmware has zero issues with this board.

DeeKey, 2020-07-14:
@dirkmueller was after this case. Together with him we tried to fix the issue with these boards. But seems that they appeared once again...