opendata-stuttgart/sensors-software #744: SPS30 continuously in Measurement-Mode - reducing lifetime

nagyrobi, 2020-06-27:
Hello team! Thank you for this wonderful solution!

I am running a Sensirion SPS30 with firmware NRZ-2020-130-B6/EN.
I've noticed that SPS30 is running fan continuously after starting the unit. This is problematic on long term, because it means according to the [datasheet](https://www.sensirion.com/fileadmin/user_upload/customers/sensirion/Dokumente/9.6_Particulate_Matter/Datasheets/Sensirion_PM_Sensors_SPS30_Datasheet.pdf) that all electronics switched on / max. power consumption, laser and fan running constantly, and it provides measured values every second. This will degrade lifetime of the sensor dramatically.

The default 145 seconds measurement interval doesn't require the sensor to run continuously, you take only one measurement and throw away other 144. It has also the adverse artifact that [after some time](https://github.com/paulvha/sps30/issues/3) the measure values for PM4 and PM10 will be errorneous.

[I see in the code](https://github.com/opendata-stuttgart/sensors-software/blob/88d3f5006c51f1f4765204e2519114a7c02aa0d2/airrohr-firmware/sps30_i2c.cpp#L95) that the command to enter in idle mode is commented out. This is incorrect. 

The sensor should stop measuring and enter idle mode after the measurement, and when a new measurement is needed, send the command to start the measurement.
I would  suggest to enter idle mode with the [device reset](https://github.com/opendata-stuttgart/sensors-software/blob/88d3f5006c51f1f4765204e2519114a7c02aa0d2/airrohr-firmware/sps30_i2c.cpp#L51) command (after power up, the sensor goes to idle mode) as that ensures of clearing data buffers also and perform clean measurements

hbitter, 2020-07-17:
Hi nagyrobi,

you mentioned the datasheet of the SPS30.

1. On page 2 of the datasheet it says that the "Typical start-up time" for "number concentration" in the range "50 – 100 #/cm3" is 30s, additionally with footnote 5: "5 Time after starting Measurement-Mode, until a stable measurement is obtained." That means a stable measurement is only obtained after 30 seconds.

2. On page 2 of the datasheet it also says about the lifetime: "Lifetime 24 h/day operation   > 10  years". That is a pretty good value (espaciallly compared to the SDS011). Though there seems to be no need not to use continous operation. By the way, it was recommended by Sensirion.

3. To run the sensor continuously you state: "It has also the adverse artifact that after some time the measure values for PM4 and PM10 will be errorneous." Why? Some more explanation why you think so would be helpfull.
First the Sensirion SPS30 does not measure PM4 and PM10 at all. PM4 and PM10 values are calculated by the SPS30. Footnote 4 "PM4 and PM10 output values are calculated based on distribution profile of all measured particles." (footnote 4, page 2 datasheet). 
The behaviour of the sensor (small or no spread between PM2.5 and PM4/PM10) is sometimes not understandable and I think Sensirion should give some more detailed explanation about calculation of PM4 and PM10 and the small spread in the datasheet.
So far  nothing indicates that continous measurement is errornous

Regards,
Helmut

hbitter, 2020-07-18:
Hi nagyrobi,

you mentioned the datasheet of the SPS30.

1. On page 2 of the datasheet it says that the "Typical start-up time" for "number concentration" in the range "50 – 100 #/cm3" is 30s, additionally with footnote 5: "5 Time after starting Measurement-Mode, until a stable measurement is obtained." That means a stable measurement is only obtained after 30 seconds.

2. On page 2 of the datasheet it also says about the lifetime: "Lifetime 24 h/day operation   > 10  years". That is a pretty good value (espaciallly compared to the SDS011). Though there seems to be no need not to use continous operation. By the way, it was recommended by Sensirion.

3. To run the sensor continuously you state: "It has also the adverse artifact that after some time the measure values for PM4 and PM10 will be errorneous." Why? Some more explanation why you think so would be helpfull.
First the Sensirion SPS30 does not measure PM4 and PM10 at all. PM4 and PM10 values are calculated by the SPS30. Footnote 4 "PM4 and PM10 output values are calculated based on distribution profile of all measured particles." (footnote 4, page 2 datasheet). 
The behaviour of the sensor (small or no spread between PM2.5 and PM4/PM10) is sometimes not understandable and I think Sensirion should give some more detailed explanation about calculation of PM4 and PM10 and the small spread in the datasheet.
So far nothing indicates that continous measurement is errornous.
Values from the SPS30 are taken every 11 sec and the arithmetic mean is done after the 145 interval.

Regards,
Helmut