opendata-stuttgart/sensors-software #746: Fan on SDS011 continues to blow

Frankster-NL, 2020-07-15:
Am running NRZ-2020-129 on a NodeMCUv3 with a BME280 and a SDS011 attached. I have set the Measuring interval (sec) to 600.

Data is indeed coming in about every 10 minutes, but if I listen closely to the SDS011 the fan keeps on spinning between measurements. Restarting the sensor and even powering it down does not resolve the issue.

What can be wrong here?

bertrik, 2020-07-15:
Perhaps the data line from the ESP8266 (TX) to the SDS011 (RX) is no longer connected correctly. This would result in the ESP no longer being able to send commands to the SDS011 (to turn the fan on or off), but still able to *receive* data from the SDS011. It's probably now stuck in the "fan on" state since that was the last received command.

Frankster-NL, 2020-08-10:
I, finally, had a look at the log and the issue appears to be due to the firmware polling the SDS011 every second. See below the log captured in about 30 seconds (measuring interval is set to 30 seconds)

End reading SDS011
## Sending to sensor.community - SDS011
Succeeded - api.Start reading SDS011
End reading SDS011
Start reading SDS011
End reading SDS011
Start reading SDS011
End reading SDS011
Start reading SDS011
End reading SDS011
Start reading SDS011
End reading SDS011
Start reading SDS011
End reading SDS011
Start reading SDS011
End reading SDS011
Start reading SDS011
End reading SDS011
Start reading SDS011
End reading SDS011
Start reading SDS011
End reading SDS011
Start reading SDS011
End reading SDS011
Start reading SDS011
End reading SDS011
Start reading SDS011
End reading SDS011
Start reading SDS011
End reading SDS011
Start reading SDS011
End reading SDS011
Start reading SDS011
End reading SDS011
Start reading SDS011
End reading SDS011
Start reading SDS011
End reading SDS011
Start reading SDS011
End reading SDS011
Start reading SDS011
End reading SDS011
Start reading SDS011
End reading SDS011
Start reading SDS011
End reading SDS011
Start reading SDS011
End reading SDS011
Start reading SDS011
End reading SDS011
Start reading SDS011
PM10 (sec.) : 13.80
PM2.5 (sec.): 10.50
End reading SDS011
Start reading SDS011
PM10 (sec.) : 14.00
PM2.5 (sec.): 10.80
End reading SDS011
Start reading SDS011
PM10 (sec.) : 13.40
PM2.5 (sec.): 10.60
End reading SDS011
Start reading SDS011
PM10 (sec.) : 16.30
PM2.5 (sec.): 10.70
End reading SDS011
Start reading SDS011
PM10 (sec.) : 15.90
PM2.5 (sec.): 10.70
End reading SDS011
Start reading SDS011
PM10 (sec.) : 15.40
PM2.5 (sec.): 10.60
PM10:  14.77
PM2.5: 10.65
----
End reading SDS011
## Sending to sensor.community - SDS011
Start reading BMP/E280
Temperature (°C): 27.93
Pressure (hPa): 101565.03
Humidity (%): 54.69
----
End reading BMP/E280
## Sending to senStart reading SDS011
End reading SDS011
Start reading SDS011
End reading SDS011
Start reading SDS011
End reading SDS011
Start reading SDS011
End reading SDS011
Start reading SDS011
End reading SDS011
Start reading SDS011
End reading SDS011
Start reading SDS011

zoomx, 2020-09-02:
I have tha same issue

dirkmueller, 2020-09-02:
can you please test with "use beta firmware" and report if the problem remains? if so, please post a screenshot of the /status page.

liutikas, 2020-09-03:
I also noticed the same issue when I enabled Debug max info. It seems to show starting and ending of the measurement

Screenshot of about 10 seconds worth of logging
[image]

zoomx, 2020-09-03:
I resolved but I don't know how exactly
I connected the sensor to the serial USB adapter and opened Realterm
The sensor was sending continuously data about one per second, string like this one (taken from the manual)
AA C0 D4 04 3A 0A A1 60 1D AB
where A1 60 is the sensor ID
Then I put the stop command, taken from the manual, in Realterm
AA B4 06 01 00 00 00 00 00 00 00 00 00 00 00 A1 60 08 AB
and pushed Send as Hex.
I used also this command
AA B4 06 01 00 00 00 00 00 00 00 00 00 00 00 FF FF 08 AB
where ID is FF FF, that is a command that should works for every sensor connected.
I noticed nothing but I was so stupid that the sensor was near my laptop fan, so maybe it worked and I didn't noticed it
Then I reconnected the sensor to the Wemos (I use a wemos instead of a NodeMcu) because I wanted to test the example of this library
https://github.com/ricki-z/SDS011
adding the stop and wake commands in the example but the library has an error that I wasn't able to correct.
But at this point I noticed that the noise from the sensor fan was different and discovered that now it starts and stops.
Unfortunately I was not able to find what was wrong and what action I did to fix that error. MAybe was simply disconnecting and reconnecting wires. But now I am shure that the sensor is not defective.

I believe that maybe a sinple sketch to test sensor start and stop may be useful, checking also if serial communications are correct.

dirkmueller, 2020-09-03:
so to explain this a bit more. when you set the debug to "max info" it prints those "start reading/end reading" lines, but that means that it just *enters* the callback routine. it doesn't mean it actually *does* something in the callback routine. these are just log statements that it enters and exits the routine. those are not logs of the actual sensor communication. 

I agree with you that it would be more useful to log the sensor communication, but the codebase is the way it is so far. If you're willing to try out test builds from me I'll be happy to provide one that helps tracing the issue down. 

you can see that it sends the command here: 

https://github.com/opendata-stuttgart/sensors-software/blob/beta/airrohr-firmware/utils.cpp#L422-L424

basically the hex bytes that it will send are exactly what you described. 

regarding "it is sometimes difficult to hear whether the sensor is on or off", there is a blinking led in the back of the sensor that blinks whenever the laser is taking a measurement. when the sensor si in measurement mode (started) it should blink once a second, and when it is sleeping (between the measurement intervals) it should not blink. thats usually easier to see than hearing fan noises in a not-so-quiet environment. 

last, but not least, the code recently was changed to not turn off the fan at all if the measurement interval is below 25 seconds, because warmup time would otherwise be too long. 

I need to improve debug output but basically that might be an error cause that the measurement interval is too short.

dirkmueller, 2020-09-03:
also, one thing that I've noticed but couldn't reproduce is that the "SDS011" sensor isn't always able to process commands. it is a guess (that I couldn't validate) that when you happen to send a command while it is currently busy trying to do something else (like processing a measurement picture or sending a reply) it loses the command. 

we currently do *not* implement 'wait for acknowledgement'  on the command so far. it was my longer term goal to rework the sensor code to retry-resend the commands until the right acknowledgement arrives, so that we have not accidental dropouts that are not covered. 

hwoever, it would be really *weird* to see these accidental misses, which in my observation happen  maybe once a day, happen for you every time for some reason.

dirkmueller, 2020-09-03:
also, please provide a screenshot of the /status page. if the SDS011 sensor doesn't show a version number there, we know that the sensor communication is broken. the sensor boots with fan on, and we send a "off" command while probing for the version number during power on self-test.

zoomx, 2020-09-03:
I read also here in the issues
https://github.com/ricki-z/SDS011/issues/10#issuecomment-401619387
that the sensor seems to not recognize serial command while doing some other things, maybe printing data.

I have the SDS version number
SDS011 | 18-11-16(4a9b)
But I don't remember if before I had or not it.

Frankster-NL, 2020-09-03:
Only thing that I can confirm is that since I moved to esphome (without even touching the connectors of the sds11) the fan stopped spinning continuously.

In the Luftdaten firmware measurement interval was set to 900 seconds (in case you want to know: in esphome it is set at 120)