opendata-stuttgart/sensors-software #747: DHT21 AM2301 sensor

WalterPepeka, 2020-07-19:
Hi everyone,
I am trying to connect a DHT21 AM2301 sensor (temperature / humidity) using same connection as the DHT22

But I get no data... Need another pin or firmware?

Hardware : NodeMCU V2
PM 2.5/10 : SDS011
Firmware : NRZ-2020-129

Regards
WP

DeeKey, 2020-07-20:
The DHT22 sensor is originally an Indoor sensor. For outdoor use it appears to be rather sensitive to water condensation after 100% rel.Humidity that keep it for very long time (sometimes forever) at 99.9% value. Also it appears to be sensitive to high UV light, which tends to cause the sensor to crash until hard power-loss restarted.

Better experiences have been made with a BME280 or SHT3x sensor, so consider those instead.

Check wiring - no loose wires
DHT22

    Pin 1 => 3V3
    Pin 2 => Pin D7 (GPIO13)
    Pin 3 => unused
    Pin 4 => GND

ricki-z, 2020-07-20:
The DHT21 isn't supported by our firmware, only the DHT22.

WalterPepeka, 2020-07-21:
> The DHT21 isn't supported by our firmware, only the DHT22.

Despite this, the sensor is still able to take data (it seems that the AM2301 works like a DHT22 ...)

ricki-z, 2020-09-02:
@WalterPepeka As mentioned our firmware only supports the DHT22.
Can we (or better you) close this issue?

closed 2020-09-02 by WalterPepeka