opendata-stuttgart/sensors-software #748: Connection refused

jwende, 2020-07-19:
Ich bekomme immer diese Information auf meinem Sensor:

Data Send | 2
Data Send Return | connection refused

Woran liegt das ?
Danke
[image]

WalterPepeka, 2020-07-19:
Hi, same to me.
It seems a server problem (MapSensor Community), data transfered to Grafana are ok

jwende, 2020-07-19:
Similar on my side. Local data and remote database ok - But Sensor is not visible on the map.
J.

JoergWende, 2020-07-20:
I can confirm that the error disappears if I uncheck API Sensor.Community 
J.

DeeKey, 2020-07-20:
Seems that this was a temporally sensor.com server  problem. Now it is resolved.
Please close the issue if working now!

JoergWende, 2020-07-20:
Not really - problem still exists

kungfo0, 2020-07-20:
Same here. OpenSenseMap works fine but API Sensor.Community produces "Data Send Return | connection refused"

DeeKey, 2020-07-20:
sensor.com server/hoster problems (((

WalterPepeka, 2020-07-21:
it might be a good idea to include a server health status page, like here
[https://setiathome.berkeley.edu/show_server_status.php](url)

ricki-z, 2020-07-29:
we are working on a server health page. At the moment it shows to many false positives as the connections within the data center are very unstable. So the page is sometimes showing problems but the everything is working when accessed from outside.

MikeMoltoHAJ, 2020-08-26:
hi, have the same problem. May be a config error? Tried any versions of config but sorry, I am not a specialist. 
[image]
[image]

marcauberer, 2020-08-26:
You could try to configure it like this:

Server: api.thingspeak.com
Pfad: /update...
...

This might help

MikeMoltoHAJ, 2020-08-27:
Hi, thanks for response. I tried this out but no help. A other question is - shall i use HTTPS or not? And what are the right ports in any case?

MikeMoltoHAJ, 2020-08-27:
#marcauberer thanks, your hint now works. I changed Port to 443. Next problem: how can i send sensor data.
e.g.  &field1=1   sent 1,   &field1=test   sent test, also with ""

ricki-z, 2020-09-02:
@jwende can we close this issue? API should be back to normal now.
@MikeMoltoHAJ if the problem persists please open a new issue as you problem is unrelated to the original problem (API Sensor.Community slow or unreachable).

JoergWende, 2020-09-02:
I still see the 'connection refused ' in the device log (web page) - but the values are on the web site . So it is q question of beauty :)
J.

dirkmueller, 2020-09-02:
the error message is supposed to be sticky. basically it shows the error cause the last time an error happened, even if it resumed to working. basically you need to compare "total measurements" with "number of errors" or basically check if "number of errors" go up over time.

any suggestions on how to make that more obvious appreciated. I don't want to wipe the errors at regular intervals, but maybe a "time of last error" field needs to be added?

JoergWende, 2020-09-03:
[image]
does this help ?

JoergWende, 2020-09-03:
ok - change on my side after restart of the sensor - no error listed. 
So it would make sense to have an error date / timestamp to see if it was a temporary issue only.
J.

dirkmueller, 2020-09-16:
To clarify how to read the status page: you do have NTP sync working (see the 8 cases of successful ntp sync). also, you had 3 out of 175 api submissions fail (which is ~ 1.7% failure rate) and the error of the last sending was "connection refused". 

so I'd say this is pretty normal and everything looks good. I need to think about a way to make this more readable (it was meant to make sense for me, but I realize that more people look at the status page :-) ).

ricki-z, 2020-09-16:
At the moment the main database server is down because of a hardware problem. Restore of nearly 1TByte needs some time.