opendata-stuttgart/sensors-software #749: DNMS I2C Bus stops working if OLED is powered of

erberj, 2020-07-21:
I created a mobile, battery powered version of the DNMS Sensor with an additional OLED 1306 Display (second I2C slave). It works fine until I switch off the OLED for power saving reasons. 
In this case debugging shows, that the NodeMCU V3 cannot read the Noise Values from the teensy 4.0 anymore. When I switch on the OLED again it works again.

hbitter, 2020-07-23:
Hi erberj,

what Firmware are you using on the NodeMCU? I recommend NRZ-2020-130-B6.

Because I just had a display connected to my test system on the desk, I disconnected +5V from the display so the display went dark. NodeMCU and DNMS were running normally and outputting the values over the USB/Serial connection of the NodeMCU. Also reapplying the +5V worked. The display showed the values again. I repeated this a couple of times. It worked always.

The diplay I have is an Adafruit OLED SSD1306 with +5V power supply. I know there are different displays on the market, some with 3V3, which may react different.
You should have a look with a scope or a logic analyzer on the I²C lines, to see what really happens on SDA and SCL.

Regards,
Helmut

erberj, 2020-07-24:
> Hi erberj,
>
> what Firmware are you using on the NodeMCU? I recommend NRZ-2020-130-B6.
>
> Because I just had a display connected to my test system on the desk, I disconnected +5V from the display so the [...]

erberj, 2020-07-24:
Hi Helmut,
thx for your reply. the firmware version ist NRZ-2020-129. But I have auto update enabled and it says: No update. Is the version above a beta version? If yes, I will try it.
This is my OLED 1306: https://www.bastelgarage.ch/oled-display-weiss-i2c-128x64-0-96?search=oled%201306. It can handle 3.3V and 5V. I use 5V (battery 3.7 ist transformed up to 5V)
I have 2 pullup resistors on the I2C bus, otherwise it wouldnt have worked reliably.
Regards
Jakob

erberj, 2020-07-24:
I have now firmware NRZ-2020-130-B6. It still does not work. I think, next try will be with an adafruit oled.

hbitter, 2020-07-24:
Hi Jakob,
it may be a problem with the power supply. So you should check, if it works with a standard USB power supply instead of the battery supply. 
Regards,
Helmut

dirkmueller, 2020-08-31:
when you say "switch oled off", are you disabling it in the configuration of the airRohr firmware using the config webpanel or are you unpowering it by pulling a physical cable but leave the software config enabled?

erberj, 2020-09-01:
The second. I put a power switch into the vin of the oled.Regards Jakob Am 01.09.20 um 01:18 schrieb Dirk Mueller
                
            
            
                
                    Von: "Dirk Mueller" <notifications@github.com>Datum: 1. September 2020An: "opendata-stuttgart/sensors-software" <sensors-software@noreply.github.com>Cc: "State change" <state_change@noreply.github.com>,"erberj" <jakob.erber@gmail.com>Betreff: Re: [opendata-stuttgart/sensors-software] DNMS I2C Bus stops working if OLED is powered of (#749)
                
                
when you say "switch oled off", are you disabling it in the configuration of the airRohr firmware using the config webpanel or are you unpowering it by pulling a physical cable but leave the software config enabled?

[
{
"@type": "EmailMessage",
"@type": "ViewAction",
"target": "https://github.com/opendata-stuttgart/sensors-software/issues/749#issuecomment-684091898",
"url": "https://github.com/opendata-stuttgart/sensors-software/issues/749#issuecomment-684091898",
"name": "View Issue"
},
"description": "View this Issue on GitHub",
"publisher": {
"@type": "Organization",
"name": "GitHub",
"url": "https://github.com"
}
}
]

dirkmueller, 2020-09-03:
> I put a power switch into the vin of the oled

ok, great, that means this might very well be a duplicate of #735 which is already fixed. I hope that @ricki-z is pushing a new B7 soon for you to verify. I'm pretty confident it is fixed already.

hbitter, 2020-09-03:
Hi Dirk,
I have some doubts that this is a duplicate of #735. DNMS has no problem with I²C clock of 500kHz. But the problem here is, that after switching the power for the display again on, the display is not working again.
As I wrote, I did a test (but with normal USB power supply and not a battery supply) and switched the power of the display on and off. The display always worked.
I would see two possibilities:
- problem with the battery power supply, thats why I asked Jakob to test it with a normal USB power supply
- it always is possible that his display controller (even if it is named SSD1306) reacts different after switching the power on and off.
Regards,
Helmut

dirkmueller, 2020-09-03:
I see. the original description is that the DNMS reading failed when the display is off, not that the display doesn't go back on when powering it on again (thats at least how I read the first comment).