opendata-stuttgart/sensors-software #750: Wrong value for BMP280 pressure in JSON value

klein0r, 2020-07-24:
Example JSON output:

```
{"software_version": "NRZ-2020-129", "age":"23", "sensordatavalues":[{"value_type":"BME280_temperature","value":"21.49"},{"value_type":"BME280_pressure","value":"100570.50"},{"value_type":"BME280_humidity","value":"55.80"},{"value_type":"DNMS_noise_LAeq","value":"43.62"},{"value_type":"DNMS_noise_LA_min","value":"40.13"},{"value_type":"DNMS_noise_LA_max","value":"53.47"},{"value_type":"samples","value":"4345430"},{"value_type":"min_micro","value":"32"},{"value_type":"max_micro","value":"120456"},{"value_type":"signal","value":"-85"}]}
```

BME280_pressure is 100570.50

But in the HTML frontend the value is correct (since it was be devided by 100 before).

https://github.com/opendata-stuttgart/sensors-software/blob/8a936e1961fc31ba4a0e086e144b2dd6e01b8990/airrohr-firmware/airrohr-firmware.ino#L1838

Please add the division by 100 also to

https://github.com/opendata-stuttgart/sensors-software/blob/8a936e1961fc31ba4a0e086e144b2dd6e01b8990/airrohr-firmware/airrohr-firmware.ino#L2710-L2718

ricki-z, 2020-07-29:
There are some applications using this format (i.e. plugins for home automation systems). So we can't really change this. You could also add this division to your broker, the result should be the same.

klein0r, 2020-07-30:
Thanks - is there is list of units used in the API? Maybe you can add another attribute to the object with the unit in the future.

closed 2020-07-30 by klein0r