opendata-stuttgart/sensors-software #751: Json message with result data

RikDrabs, 2020-07-25:
The server distributing the Json messages with result data is down. 
( http://data.sensor.community/airrohr/v1/sensor/17546/ )
The last couple of weeks this server worked irregularily. Is there a problem of overload here too?
How many messages per minute is the limit of the server? For one user? For all users?
The correct functioning of my display, and the display of others, depend on these Json messages.
Are there other places, other methods, to get the same information (maybe .CSV files?) as alternative, when the server is down?
Thanks for the good work!

marcauberer, 2020-07-25:
A problem on the hosters side could be jointly responsible for these issues. I know that Luftdaten / Sensor.community host their data on Strato servers (common German hosting provider), which have experienced some issues the last couple of weeks, more information [here](https://www.heise.de/news/Performance-Probleme-bei-Strato-nach-Software-Updates-4850607.html) (German article). The Feinstaub-App Servers are hosted on Strato servers as well and we are also experiencing some problems with the server performance.
We're currently restrucuring our backend system towards a "cross-hoster" solution with a more modern and high-availability architecture. I hope, that will reduce those problems in the future ;)

knalla66, 2020-07-27:
I have the same Problems with my Sensors- for example:
http://data.sensor.community/airrohr/v1/sensor/41803/

is there any expectation when the hoster fix their performance issue?

I fetch several sensors with Node-Red over this link http://data.sensor.community/airrohr/v1/sensor/sensorID to visualize in a dashboard and store it in InfluxDB for further analyzation. It is not working since 23.7. 18:00 anymore, which is quit long.  I know that you rely on the hoster but is there any workaround or other access wich is functional? would appreciate any help, thx.

ricki-z, 2020-07-29:
We had to shutdown the data.sensor.community server for a while. There seems to be some sites (including our map) and tools and apps that are generating too many request in a short time.
The last days I had to split the requests over more servers. At the moment the servers are up again.
If you don't need the data in real time you could also download the data at archive.sensor.community.

knalla66, 2020-07-31:
Servers are running now, thx.
Thanks for your hard work and providing a fast solution, keep up your work!

FH JOANNEUM Gesellschaft mbH
Rechtsform/Legal form: GmbH
Sitz: Graz
Firmenbuchgericht/Court of registry: Landesgericht für ZRS Graz
Firmenbuchnummer/Company registration: FN 125888 f
DVR: 0813559
UID-Nr.: ATU 42361001
https://www.fh-joanneum.at/impressum

dirkmueller, 2020-08-31:
Can this issue be closed?

closed 2020-10-18 by ricki-z