opendata-stuttgart/sensors-software #752: Feature request: Auto-Restart

Henner7, 2020-07-26:
In a weak Wifi the connection gets frequently lost. In my usecase the sensor does not reconnect to Wifi automatically again. Usually I get an email reminder after one day and then I need to re-start the sensor manually by disconnecting the power supply.

**Feature request:**
Add an option in the firmware: "Restart sensor each xxx hours", allow the user to set value between 0...999.

**Expected behaviour:**
When the restart value is set to 0, not automated restart is done. Otherwise the sensor is restarted after the specified number of hours.

irukard, 2020-07-31:
In NAMF (nettigo/namf) we implemented network watchdog which restart sensor if three 10-ping tests fail. https://github.com/nettigo/namf/tree/master/src/sensors/network_watchdog

dirkmueller, 2020-08-31:
@Henner7 well, it is supposed to automatically reconnect on wifi loss. any chance you can capture the serial output while this happens? would be great for tracing this down..

irukard, 2020-09-01:
Hint: Sensors might struggle with WPA2 implementation in MikroTik routers.

ricki-z, 2020-09-02:
@irukard this isn't related to this feature request. If this is an issue please open a new one.

Henner7, 2020-09-03:
> @Henner7 well, it is supposed to automatically reconnect on wifi loss. any chance you can capture the serial output while this happens? would be great for tracing this down..

Good morning, currently it disconnects every day. The reconnection can be re-established by either re-starting the sensor or by re-starting the Wifi. This is really strange. Unfortunately I cannot connect the serial output as long the sensor is mounted outside in the garden. You see that my original request is not about automatic reconnection but it is about timed re-start of the sensor indest.

dirkmueller, 2020-09-03:
Technically the firmware already automatically restarts every 28 days, this is currently hard coded and cannot be changed. I was however thinking that instead of time based restart the wlan-resume behavior could be improved instead, as with a time based restart you will still be losing measurements until the time based restart hits, which is unwanted.

Also the firmware does on boot a self test off the sensors which increases wear. So a time restart with very short intervals is not useful. I hope you understand that as trying to solve the root cause might actually be better in the end.

closed 2020-09-04 by Henner7