opendata-stuttgart/sensors-software #754: Pa Werte

oe3osb, 2020-08-05:
Hallo,
mir ist aufgefallen, dass die Luftdruckwerte vom BME280, die an opensensemap.org gesendet werden, abweichend zu den auf  devices.sensor.community sind.

**opensense:**
[image]

**devices.sensor.community:**
[image]

Vielen Dank fuer die Mühe!
Erich

imaluschi, 2020-08-17:
Da scheint auf devices.sensor.community ein "h" für Hektopascal bei der Anzeige der Einheit zu fehlen. Luftdruck auf Meereshöhe in diesem Fall 101873 Pa = 1018.73 hPa = 1018.73mbar = 1.01873 bar.
Korrekte Einheit bei opensensemap, allerdings ungewohnt zum Lesen.

dirkmueller, 2020-09-02:
The issue seems to be fixed? I see this on my devices.sensor.community page:

```
Luftdruck: | 984.85 hPa
-- | --
Luftdruck auf Meereshöhe: | 1020.76 hPa
-- | --
```

Please reopen if you still see an issue.

closed 2020-09-02 by dirkmueller

ricki-z, 2020-09-02:
@oe3osb Since today the site devices.sensor.community is showing the correct unit. I had to find my test device first to verify this ;-)