opendata-stuttgart/sensors-software #756: Missing values in data.json

Bouni, 2020-08-17:
Hi, 

I use the /data.json endpoint to get the sensor data into my Home-Assistant setup like explained [here](https://github.com/opendata-stuttgart/sensors-software/issues/405#issuecomment-569447561).

This works fine most of the time, but for some reason my SDS011 sometimes stops giving values (which i have to look into but thats not the point here, a power cycle solves the problem).

The data.json endpoint returns an array with values, but if a value is missing it simply vanisches from the array.
This causes the index to shift and my readings to be wrong.
I would highly prefer if the value instead would be set to null.

Bouni, 2020-08-20:
On my sensor at home I get following data:

```
curl http://feinstaubsensor.home/data.json
{"software_version": "NRZ-2020-129", "age":"18", "sensordatavalues":[{"value_type":"temperature","value":"24.40"},{"value_type":"humidity","value":"56.90"},{"value_type":"samples","value":"4355246"},{"value_type":"min_micro","value":"32"},{"value_type":"max_micro","value":"19966"},{"value_type":"signal","value":"-73"}]}
```
In my Hackerspace, I get the full set of data (but i have the same issue there from time to time):

curl http://feinstaubsensor-14951931.bdstr.reaktor23.org/data.json
{"software_version": "NRZ-2020-129", "age":"188", "sensordatavalues":[{"value_type":"SDS_P1","value":"5.75"},{"value_type":"SDS_P2","value":"2.45"},{"value_type":"temperature","value":"26.40"},{"value_type":"humidity","value":"43.80"},{"value_type":"samples","value":"7235625"},{"value_type":"min_micro","value":"32"},{"value_type":"max_micro","value":"20044"},{"value_type":"signal","value":"-44"}]}

Bouni, 2020-08-26:
I looked a bit into the actual firmware.
You call `add_Value2Json` in each function that reads sensor data but only if the read was successful.
Wouldn't it make more sense to call the function `add_Value2Json` anyway but with `null` as value?

Maybe somthing like this in the else branch (obviously this should be done for all reading functions)
```
add_Value2Json(s, F("SDS_P1"), F("null"));
add_Value2Json(s, F("SDS_P2"), F("null"));
```

That way the structure of the JSON would be consistent even if a reading fails!

dirkmueller, 2020-08-31:
Thats true, but the json is also sent to the api data collectors, and we don't want to send "garbage" to the world. a missing value basically means it was disabled/reading failed.

isn't there a way to make your parsing routines more robust? if you are afraid of using a json library, maybe the prometheus endpoint, which provides  nice line-based protocol is easier for you to integrate?

Bouni, 2020-09-01:
I unserstand that you don't want to send garbage, but to be honest, it would make a lot more sense to sent the value as null and not simply drop it, right?
A failed reading or a missing sensor is not "garbage" in my opinion. 
What if you send data as an object rather than an array. That way one could simply check for the key instead on relying on the array index.
This would be ideal in my opinion, in case of failure or missing sensor, keep the key but set the value to null :
```
{
    "software_version": "NRZ-2020-129", 
    "age":"30", 
    "sensordatavalues": {
        "SDS_P1": {"value":"1.95"},
        "SDS_P2": {"value":"0.80"},
        "temperature": {"value":"20.50"},
        "humidity": {"value":"57.40"},
        "samples": {"value":"4376966"},
        "min_micro": {"value":"32"},
        "max_micro": {"value":"20138"},
        "signal": {"value":"-75"}
    }
}
```

In the current state I need to get the expected array index and then check if the type matches what I expect at this index. Or loop over the array until I find the desired type.

I guess you're not willing to change that as it breaks all parties using the current state?

Home Assistant does have a prometheus integration, but that does just expose metrics.

May be I could loop over the array to get the values if they are present 🤔

Bouni, 2020-09-01:
I found a [custom component for Home Assistant](https://github.com/lichtteil/local_luftdaten) that seems to do exactly what I said before, it loops over the sensor types and checks for the desired one.

So I guess this solves my problem.

closed 2020-09-01 by Bouni