opendata-stuttgart/sensors-software #759: SDS PM-Sensor not working with NRZ-2020-130-B6

imaluschi, 2020-08-24:
When trying the last beta NRZ-2020-130-B6 (because of using DNMS), i realized that the SDS is not working with this Firmware.

Screenshot with NRZ-2020-130-B6:
[image]
Checking cables and all stuff, no chance to get SDS working. Then i tried to upload the last (non-beta) version NRZ-2020-129 - and it worked.
Screenshot with NRZ-2020-129:
[image]

Furthermore, to compile (and upload) the beta, some lines had to be commented, "PIN_CS" seems to be defined nowhere (at least for ESP8266).
//        digitalWrite(PIN_CS, HIGH);
//        delay(500);
//        digitalWrite(PIN_CS, LOW);

[image]

Otherwise nice work, i like the new Config page ;-)

imaluschi, 2020-08-24:
Sorry, i realized  #737 is the same issue.
For the questions:
Yes, cabling is ok, SDS is working with NRZ-2020-129 but not working with NRZ-2020-130-B6, only firmware-update without touching the device/cables ;-)

dirkmueller, 2020-09-02:
closing as duplicate to #737.

dies war eine regression von #769 . es wurden global parameter geändert die dazu geführt haben dass die kommunikation mit dem SDS nur noch mit 115200 baud läuft. was mit der alten arduinocore version nicht geht.

das problem ist bereits gefixed.

closed 2020-09-02 by dirkmueller