opendata-stuttgart/sensors-software #760: "An eigene API senden"    ThinkSpeak  - funktioniert nicht

MikeMoltoHAJ, 2020-08-26:
Hallo - so sehr ich hier auch suche - ich finde keinen Weg meine Daten vom
Feinstaubsensor
Firmware: NRZ-2020-129 
an ThinkSeak zu senden. Hat vielleicht jemand die korrekten Einträge für die Konfig? 
Bin leider kein Fachmann, habe diverse Einstellungen mehr auf gut Glück versucht, es kommt aber nichts an.

MikeMoltoHAJ, 2020-08-27:
[image]
[image]

ricki-z, 2020-08-27:
Unsere Firmware unterstützt ThingSpeak nicht. Wir müssten für jede dieser Plattformen einen extra Format-String abspeichern und dann beim Senden entsprechend mit Werten füllen. Und das auch noch abhängig von den aktiven Sensoren. Bei dem verfügbaren Speicher ist das dann etwas viel verlangt.
Wer Daten an eine andere als die implementierten APIs senden will, kann "An eigene API senden" nutzen, um die Daten an ein Skript zu senden, welches die Daten dann in das benötigte Format umwandelt und dann an z.B. ThingSpeak weiterleitet.
Ein Beispiel-Skript gibt es unter https://github.com/opendata-stuttgart/madavi-api/blob/master/data_simple.php

MikeMoltoHAJ, 2020-08-28:
Danke für die Antwort. Die Fehlermeldung war weg nachdem ich den Serverstring aufgeteilt habe und ab /update?... in die Pfad Zeile geschrieben habe. Port 443 . Dann werde ich mich im Winter hinsetzen müssen und PHP und Javascript lernen. z.Z keine Ahnung

closed 2020-08-28 by MikeMoltoHAJ