opendata-stuttgart/sensors-software #780: Sensor running beta unable to upload results

liutikas, 2020-09-03:
I have two sensors set up, one running stable `NRZ-2020-129` and one running beta `NRZ-2020-130-B6/EN (May 8 2020 11:25:44)` . The sensor running stable software works great, but `esp8266-564214` running beta software used to work a few weeks back, but it seems unable to upload to sensor.community. Sensors does measure, going to the local URL I can see sensor data for PM2.5 and PM10. When I go to `Device status page` I see the number of measurements and it is equal to failed data upload attempts with an error `connection refused`. I'm not sure how to proceed in debugging this further. Do you have any suggestions?

liutikas, 2020-09-03:
Potentially a duplicate of https://github.com/opendata-stuttgart/sensors-software/issues/748 but keeping it separate as it is only affecting the sensor running beta software

dirkmueller, 2020-09-03:
@liutikas please post a screenshot of the api configuration tab and also of the (debug level) page after the output stabilizes. Currently the serial output capturing is a bit slow. Thanks in advance.

liutikas, 2020-09-03:
[image]
[image]
[image]

liutikas, 2020-09-03:
It seems that it is unable to reach ntp and so the time is wrong likely causing connection refusal. Is beta using different logic for ntp?

dirkmueller, 2020-09-03:
yes, there is one commit difference between master and beta: 

https://github.com/opendata-stuttgart/sensors-software/commit/a99601592e33c71f2a78f424e8c646408cec4fc8

I'd be very surprised if reverting that commit would fix ntp sync for you. 

also, as you have SSL turned off, the incorrect system time shouldn't be fatal for sending. also the error wouldn't be "connection refused" then. 

is it possible that your wifi accesspoint/internet router is blocking the client from internet access? that would explain the lack of NTP and the tcp connection refused. we can also see that OTA is failing. I did change the DHCP hostname between master and beta, so if your firewall rules are based on the DHCP hostname, then that changed with beta.

liutikas, 2020-09-04:
I was able to get to that disabling ipv6 on my router makes the beta sensor work again. It seems that something is going wrong when making network requests using ipv6. I am not seeing this issue on any other devices on my network. I would really like to enable it back again. How can we debug this further?

dirkmueller, 2020-09-04:
Interesting observation, thanks for debugging this!

for me the ntp servers that are used (0.pool.ntp.org and 1.pool.ntp.org) are not handing out ipv6 addresses, so it is surprising that ipv6 has anything to do with it. is this different in your case? or is it also only giving out A records, with other words only ipv4 addresses ?

is your router doing dual ipv4+ipv6 or ipv6-only? I don't have a working ipv6-only setup at the  moment for reproducing this. 

as a temporary measurement, because we know that ipv6 support hasn't been well tested, we have disabled the ipv6 support again so it will not affect further users. need to fix this properly however.

dirkmueller, 2020-09-15:
@liutikas does this work again with -B9 ?

liutikas, 2020-09-16:
It seems to be able to connect it NTP, but unable to upload data to sensors.community

[image]

liutikas, 2020-09-16:
It seems like it does not work even with ipv6 disabled now :/ I'll have to dig in more and see what's changed after the OTA

[image]

dirkmueller, 2020-09-16:
I think the current upload issues are a server side issue, I can't even curl https://api.sensor.community/ anymore, it just hangs.  /@ricki-z

ricki-z, 2020-09-16:
The main database server is down because of a hardware failure. Restore may need some time.

liutikas, 2020-09-16:
Would it be possible to have a service status somewhere on sensor.community so these kind of outages would be obvious

ricki-z, 2020-09-17:
A service status page is work in progress. It needs some more finetuning (showing failures that are irrelevant for users).