opendata-stuttgart/sensors-software #781: Sending GPS-Data to InfluxDB

KoffeinKaio, 2020-09-04:
__Steps to reproduce:__
List the minimal actions needed to reproduce the behavior.

1. Connect NEO 6M GPS Reciever to Airohr
2. Setup the Sensor to send to InfluxDB
3. ???
4. Profit! ... or maybe not

__Expected behavior:__
Data is successfully sent to influxdb

__Actual behavior:__
InfluxDB replies with an error.
Error only happens when I activate the GPS Reciever.

__Environment info:__

* System info: Firmware NRZ-2020-130-B7/DE (Sep 4 2020 11:59:14) on NodeMCU 2
* InfluxDB version: InfluxDB v1.8.2 (git: 1.8 6a1299e8c6c81ef8fd573ed2b700217121baf04e)
* Other relevant environment details: Container runtime, disk info, etc

__Logs:__
`
{"error":"unable to parse 'feinstaub,node=esp8266-xxxxxx SDS_P1=6.62,SDS_P2=2.58,BME280_temperature=22.61,BME280_pressure=99671.41,BME280_humidity=61.18,GPS_lat=11.11111,GPS_lon=11.111111,GPS_height=260.00,GPS_date=09/04/2020,GPS_time=19:41:47.00,samples=1097970,min_micro=48,max_micro=4399353,signal=-73': invalid number"}
`

ricki-z, 2020-09-05:
@KoffeinKaio Please test the latest beta firmware. The problem was the format for GPS_time and GPS_date (non-quoted string). The latest beta should now send both in one string GPS_datetime.

KoffeinKaio, 2020-09-07:
additional debug:

`
20:56:17.777 -> ## Sending to sensor.community - GPS
20:56:17.824 -> Request failed with error: 400
20:56:17.824 -> Details:{"detail":"JSON parse error - Expecting ',' delimiter: line 1 column 219 (char 218)"}
20:56:17.926 -> ## Sending to madavi.de: 
20:56:18.554 -> Succeeded - api-rrd.madavi.de
20:56:18.554 -> ## Sending to opensensemap: 
20:56:19.485 -> Request failed with error: 400
20:56:19.532 -> Details:{"code":"InvalidContent","message":"Invalid JSON: Unexpected number in JSON at position 466"}

`

dirkmueller, 2020-09-12:
Fix was merged, can we close this?

KoffeinKaio, 2020-09-14:
Ill test it with firmware from ota this evening and will close it if it works

KoffeinKaio, 2020-09-15:
works for me with Firmware version: NRZ-2020-130-B9/EN (Sep 8 2020)

PR #785 (2020-09-08): Fixed for sending GPS_datetime Fixes #781
Some fixes for sending GPS_datetime to influxdb and fixes for correctring the name of the measurement.

1:
According to https://github.com/opendata-stuttgart/feinstaub-api/blob/master/feinstaub/sensors/models.py#L119 GPS_datetime should be named GPS_timestamp.

2:
Also influxdb likes to get those values in quotes as string - but other APIs do not, so I used the isNumeric function mostly copied from http://tripsintech.com/arduino-isnumeric-function/ to detect a non-numeric and quote it only for influxdb.

3:
I also added a default value for datetime of  [... cut]
commit 7ab75c50 (2020-09-08): Merge pull request #785 from KoffeinKaio/gps-data-fix

Fixed for sending GPS_datetime Fixes #781
closed 2020-09-15 by KoffeinKaio