opendata-stuttgart/sensors-software #783: Humidity correction for BME280

sumerland, 2020-09-05:
I have compared the outputs of BME280 and SHT35 with data from the nearby weather station throughout the past weeks. The SHT35 is typically within 1-2% rel. humidity (taking temperature differences into account), whereas the BME280 is usually 20% lower and 2.5degC higher. The temperature error can be corrected using the temperature compensation in the configuration (luckily it does not affect the SHT output!). Could you please add a humidity correction, too?

P.S.: Yes, I did condition the BME280 as recommended in the manual and the error is similar across the batch of 5 sensors I purchased.

sumerland, 2020-09-06:
OK, I felt brave today, got visual studio code and the beta as zip file from this morning. In the bmx280 section I repeated what's done for the temperature correction and added a humidity correction. The GUI now has an additional humidity correction field.

[image]

Looking at the previous weeks the delta to the (trustworthy) SHT35 is -20% on humid days and about -10% in dry air. So I went ahead with +15% correction. It's too early to see the effect but you get the idea of the difference of the original data:

[image]

[image]

And for the sake of completeness, here is the effect of the 2.5degC correction:

[image]

It's a hack - and a trivial one, as even I managed to do it. I have no idea how to upload such a change and whether it is welcome at all. Please let me know if you need more info.

friedpa, 2020-09-06:
Hi, this "Hack" is very appriciated! I asked already for such correction fields. Can you please let me know what you chenged? Thx!

sumerland, 2020-09-06:
In airrohr-firmware.ino around line 2400 the temperature correction is applied

`last_value_BMX280_T = t + readCorrectionOffset(cfg::temp_correction);`

A few lines later the humidity is read for the BME280. There I modified the existing line to

`last_value_BME280_H = h + readCorrectionOffset(cfg::humid_correction);`

From there on you can check where `temp_correction` is defined/used and add equivalent lines for the new `humid_correction`. You also have to check the following files for `TEMP_CORRECTION` and `temp_correction` and duplicate/modify those lines for the humidity correction:

- airrohr-cfg.h
- airrohr-cfg.h.py
- ext_def.h
- defines.h
- intl_*.h

friedpa, 2020-09-06:
@sumerland Thx a lot!

dirkmueller, 2020-09-06:
it is likely that the 15% humidity offset is just a result of the 2.5 °C temperature offset, as the humidity reading of the sensor is compensated with a complex calculation based on the temperature reading of the sensor. 

According to the datasheet the absolute accuracy is +/- 1.0 °C, so 2.5 is well beyond the specified tolerance. you're saying that all batch-of-5 sensors did have the same issue?

sumerland, 2020-09-06:
I have checked that many times, albeit using an online calculator using a simplified approach ([https://rechneronline.de/barometer/luftfeuchtigkeit.php](url))

An example from yesterday 20:00:

SHT: 15,16degC, 82% rel. humidity
BME: 17,44decC, 64% rel. humidity

Feeding the SHT's values into the calculator yields an absolute humidity of 10.61g/m^3.
Using the BME's temperature this is translated into a relative humidity of 71,4% - still about 7% off the BME's reading. I don't know if a more accurate numerical approach is able to close that gap.

Yes, I got a set of five BMEs and they were all very similar. Here is a comparison after some more hours. You can clearly see when I applied the temperature and the humidity correction.

[image]

dirkmueller, 2020-09-06:
According to datasheets, SHT35 has +/- 2% RelH tolerance (with readings < 90%) and BME280 ~ +/-3% 

so this is pretty close. I would however trust the values of a SHT35 any time over a BME280 given its much much higher price and better certification. also long term stability seems to be a lot better with a SHT sensor than the BMEs, they tend to age relatively quickly.

I just wonder if we have a defect in the sensor data reading code that would explain this huge difference. I only have two BME280 and they're within < 0.2°C of each other and I'd think < 0.5 compared to other sensors (sht 32, various commercial weather/temperature stations. However, there is a lot of sensor-specific code based on calibration data that could mean that some batches expose issues that others do not.

can I give you a test firmware with some extra debugging code for testing?

Franz4596, 2020-09-06:
I noticed ~3  years ago that DHT22 humitdity values was drifting after 6 month operation against 100% (when cold, high RH), and a regenaration (drying) was not helpful. So I installed BME280's at DHT22 in parallel. I have about 20 BME280 / DHT22 in operation, see eg. https://api-rrd.madavi.de/grafana/d/q87EBfWGk/temperature-humidity-pressure?orgId=1&var-chipID=esp8266-1492670 My experiance: 
- the Rel. Humidity of most DHT22 drifts to 100%, same at BME280 but slower.
 I assume the humidity depending dielelectricum gets i a saturated state.
- the temperatures of all my DHT and BME are close together (<+-0.5°C) and represent the true t,
   (in case of a quick t-change, eg direct sun, big t-differnces (...15°C) may occur)
- sometimes a BME280 failed, and needed to be powerd off/on, some I had to be replaced.
- see eg. https://api-rrd.madavi.de/grafana/d/q87EBfWGk/temperature-humidity-pressure?orgId=1&var-chipID=esp8266-1492670 
- finally the pressure of bme280 is not the best, difference of 0...10mbar  
Nevertheless, I am very satisfied about the accuracy of temperatur data for this low cost devices which is a good because better than 1°. A temperature correction is only not needed. And a humidity corretion makes not sense, because value drifts over the time.

sumerland, 2020-09-07:
@dirkmueller I am happy to run the test firmware. I have also ordered three additional BME280 on different breakout boards from different sources. Two should arrive today and I will give them a go.

On the map I can see a sensor just a few km away which runs a BME280 and has sensible humidity readings.

sumerland, 2020-09-07:
I have just installed a different BME280 (pimorino.com, replaces the previous from AZDelivery). Temperature is not within 0.5K. Humidity is probably too early, as the sensor should undergo the conditioning procedure but straight out of the box it's now "just" 5% below the SHT35.

Franz4596, 2020-09-08:
**@summerland** Now I operate a SHT31D (shall be water fluid resistant).  If you like you can compare values of several measurements arround my house:
1. streetside house: SDS011(PM:1885), DHT22(t,H:1886), BME280(t,H,p:7067), DNMS(LAeq:38366)  esp8266-1492670
see https://api-rrd.madavi.de/grafana/d/q87EBfWGk/temperature-humidity-pressure?orgId=1&var-chipID=esp8266-1492670
2. gardenside house corner left: SDS011(PM:3141), DHT22(t,H:-), BME280(t,H,p:3142) esp8266-2281326
see https://api-rrd.madavi.de/grafana/d/q87EBfWGk/temperature-humidity-pressure?orgId=1&var-chipID=esp8266-2281326 
3. gardenside house corner right: DNMS(LAeq:49498), SHT31(t,H:49499) esp8266-14726029
see https://api-rrd.madavi.de/grafana/d/q87EBfWGk/temperature-humidity-pressure?orgId=1&var-chipID=esp8266-14726029

values 08.09.2020 04:00
1. DHT22    13.0°C / 99%     
   BME280   12.7°C / 88% / 1017hPa
2. DHT22    14.9°C / 99%     
    BME280  15.8°C / 94% / 1018hPa (7.9.10 0:00-17:00 hangup)
3. SHT31     14.3°C / 73%     
My opinion:  temperature values are in an acceptable range, rel.humidity was horrible. I do not see several °C difference like you. If the sensors are placed close togther diff. is dt<0,5K. 
The DHT22 is not good for cold, high RH outdoor-operation, it drifts to RH->100% , BME280 RH drift is better, but hang up sometimes. SHR31D is more expensive, maybe hopefully better.
Note, I tried to regenerate DHT22 by drying in SilicaGel, but with bad success.

sumerland, 2020-09-08:
@Franz4596 Not sure about your DHT22 - this was the first sensor I replaced on my board and I don't longer use it.

Here is the temperature and humidity data of the "new" BME280 vs SHT35. 

[image]

I have no more complaints and went back to the current beta software. Humidity and temperature corrections are no longer required. Of course I still don't know if the previously used BME280 was dodgy or if the firmware does not handle it properly. I am happy to send the sensor to anybody willing to debug it.

Franz4596, 2020-09-11:
Perfect, it looks that your five BMEs are a bad production. 
 I "use" DHT22 (even bad humidity) because its recomanded by luftdaten only. Always I add a BME280 because aditional pressure and a little better humidity behavior. A few BME280 hang up sometims and some got damaged. Even temperature of BME and DHT are close together, I recomand to use SHT31/35. 
Think this issue could be closed.

closed 2020-09-11 by sumerland