opendata-stuttgart/sensors-software #786: Problem with GPS and JSON in NRZ-2020-130-B9

n0c0n, 2020-09-08:
Hi,
I've downloaded beta brunch yesterday (07.09.2020) and as you can see in log (read from COM) below there is reported problem after tuning on GPS in Configuration\Sensors
**Details:{"detail":"JSON parse error - Expecting ',' delimiter: line 1 column 225 (char 224)"}**
After turning off GPS everything is OK.
```
output values to display...
PM1:   5.50
PM10:  11.50
PM2.5: 10.00
----
Lat: -200.000000
Lng: -200.000000
Date:
Time
Altitude: -1000.00
----
## Sending to sensor.community - PMSx003
Succeeded - api.sensor.community
Temperature (°C): 23.61
Pressure (hPa): 99288.16
Humidity (%): 36.35
----
## Sending to sensor.community - BMP/E280
Succeeded - api.sensor.community
## Sending to sensor.community - GPS
Request failed with error: 400
Details:{"detail":"JSON parse error - Expecting ',' delimiter: line 1 column 225 (char 224)"}
## Sending to madavi.de:
Succeeded - api-rrd.madavi.de
## Sending to Server FS App:
Succeeded - server.chillibits.com
## Sending to custom influx db:
Data read failed
Succeeded - ****.****.**
## Sending to custom api:
Succeeded - ****.****.**
Time for Sending (ms): 2198
```
PS
I also have old problem with sending data to InfluxDB when GPS is turned on. From what I've found (a few month ago) it's probably caused by differences in time format used by InfluxDB and time format sent with GPS data. I didn't managed with this problem so I've stopped using GPS ;-)

dirkmueller, 2020-09-12:
is this error before or after the fix from #781 ?

n0c0n, 2020-09-15:
As I've wrote I used beta code from 7th September and, as I see, the fix was committed by KoffeinKaio 8th September. So my compilation was before fix #781.

n0c0n, 2020-09-15:
With today's beta version (15.09.2020) problem with GPS and InfluxDB is fixed. Thanks.

But I still see in log from COM this thing as below
With GPS fix (location):
```
Lat: **.***** [hidden value]
Lng: **.***** [hidden value]
DateTime: 2020-09-15T11:20:41.000
Altitude: 367.40
----
## Sending to sensor.community - PMSx003
Succeeded - api.sensor.community
Temperature (°C): 26.12
Pressure (hPa): 99149.09
Humidity (%): 41.63
----
## Sending to sensor.community - BMP/E280
Succeeded - api.sensor.community
## Sending to sensor.community - GPS
Request failed with error: 400
Details:["sensor could not be selected."]
## Sending to madavi.de: 
```
Without GPS fix
```
## Sending to sensor.community - GPS
Request failed with error: 400
Details:{"sensordatavalues":[{},{},{},{"value":["This field may not be blank."]}]}
```
I don't know is something wrong or it should be like that.
Beside of this, for me everything is OK.

KoffeinKaio, 2020-09-15:
As far as I know sensor.community doesnt register gps-"sensor"-data anyways

n0c0n, 2020-09-15:
Oh, OK. Thanks for enlightening me.
BTW. Can I have a request?
Could we use the GPS date show in WEB in more international format like DMY not the US format MDY?
As you can see https://en.wikipedia.org/wiki/Date_format_by_country more population use DMY than MDY ;-)
Yes, I know. I can make that change in code for my own (because I know where), but what with people that don't know how?

KoffeinKaio, 2020-09-15:
Please do a separate Feature Request as an Issue for this, but this would be a really low priority enhancement.
First would be to find out if we can send any other formats to the APIs and change it in https://github.com/opendata-stuttgart/sensors-software/blob/master/airrohr-firmware/airrohr-firmware.ino#L3345 or if we have to re-format it for webui.

n0c0n, 2020-09-15:
As you said it's low priority. So lets close this issue. Problems was solved.

closed 2020-09-15 by n0c0n