opendata-stuttgart/sensors-software #787: airrohr-firmware does not compile for esp32

liutikas, 2020-09-10:
I attempted to build airrohr-firmware for `lolin d32 pro` in arduino IDE, but I am unable compile. I get the following errors:

```
airrohr-firmware/airrohr-firmware.ino: In function 'void yield_for_serial_buffer(size_t)':
airrohr-firmware:784:13: error: 'class HardwareSerial' has no member named 'perform_work'
   serialSDS.perform_work();
             ^
airrohr-firmware/airrohr-firmware.ino: In function 'String SDS_version_date()':
airrohr-firmware:797:13: error: 'class HardwareSerial' has no member named 'perform_work'
   serialSDS.perform_work();
             ^
airrohr-firmware/airrohr-firmware.ino: In function 'void webserver_status()':
airrohr-firmware:1934:21: error: 'class EspClass' has no member named 'getFullVersion'
  versionHtml += ESP.getFullVersion();
                     ^
airrohr-firmware:1938:77: error: 'class EspClass' has no member named 'getHeapFragmentation'
  add_table_row_from_value(page_content, F("Heap Fragmentation"), String(ESP.getHeapFragmentation()), "%");
                                                                             ^
airrohr-firmware:1963:64: error: 'class EspClass' has no member named 'getResetReason'
  add_table_row_from_value(page_content, F("Reset Reason"), ESP.getResetReason());
                                                                ^
airrohr-firmware/airrohr-firmware.ino: In function 'void connectWifi()':
airrohr-firmware:2368:7: error: 'class WiFiClass' has no member named 'hostname'
  WiFi.hostname(cfg::fs_ssid);
       ^
airrohr-firmware/airrohr-firmware.ino: In function 'void setup()':
airrohr-firmware:4213:12: error: 'class HardwareSerial' has no member named 'enableIntTx'
  serialSDS.enableIntTx(true);
            ^
airrohr-firmware:4238:38: error: 'class EspClass' has no member named 'checkFlashConfig'
  if ((airrohr_selftest_failed = !ESP.checkFlashConfig(true) /* after 2.7.0 update: || !ESP.checkFlashCRC() */)) {
                                      ^
```

Is this mean to work? From a quick glance serialSDS uses [`HardwareSerial`](https://github.com/espressif/arduino-esp32/blob/0607d36734d52d67be1b88764bf446fac7eb62f3/cores/esp32/HardwareSerial.h#L55) which in fact does not have `perform_work` function, the way [`SoftwareSerial`](https://github.com/plerup/espsoftwareserial/blob/master/src/SoftwareSerial.h#L199) does. Is this specific call not required for `HardwareSerial`?

Similarly, [`ESP.getFullVersion()`](https://github.com/esp8266/Arduino/blob/master/cores/esp8266/Esp.h#L119) exists for ESP8266, but not for ESP32 [`EspClass`](https://github.com/espressif/arduino-esp32/blob/master/cores/esp32/Esp.h#L61)

liutikas, 2020-09-10:
It seems that it was disabled in https://github.com/opendata-stuttgart/sensors-software/commit/9ac31454b083ff1537b06c74305fb20bfc5c7548

Does that mean that this project no longer intends to support ESP32?

peterhinson, 2020-09-10:
I was able to get the firmware working with ESP32 recently with a few small changes. The errors you posted above look like version/package issues to me (I hit entirely different errors, mostly at runtime). You might try compiling using PlatformIO + VS Code, it does a better job of managing dependencies than the Arduino IDE.

dirkmueller, 2020-09-12:
ESP32 should be supported in the beta branch. please don'tr try the master branch

peterhinson, 2020-09-13:
@dirkmueller I had to make a few changes to beta to get ESP32 working ([seen here](https://github.com/opendata-stuttgart/sensors-software/compare/beta...peterhinson:fix-esp32-issues)). It looks like `setNetworkTime()` was moved before `connectWifi()` [recently](https://github.com/opendata-stuttgart/sensors-software/commit/207356511657b5a8bfd2b9affdd05cdcda31f140#diff-08a2766b70cf0855dc046320abdaf32eR4217-R4224) which causes a runtime error on ESP32s. If this was a regression, I can submit a PR for my branch.

dirkmueller, 2020-09-15:
@peterhinson unfortunately that change was intentional, as on ESP8266 arduinocore the ntp sync callback isn't called after initial connect anymore if DHCP hands out a NTP server. so we need to call configTime() beforehand to tell core the fallback ntp servers, and then invoke DHCP (which updates ntp and calls ntp sync). 

if thats a fatal issue for esp32 we need to add a ifdef in the code unfortunately.

peterhinson, 2020-09-16:
@dirkmueller aye, I suspected there might be a good reason for the change. Unfortunately calling `configTime()` prior to a WiFi connection does indeed causes an abort on the ESP32: `.../freertos/queue.c:1442 (xQueueGenericReceive)- assert failed!` (some discussion on [this issue](https://github.com/espressif/arduino-esp32/issues/3467)). I'll wrap those calls in an ifdef and submit a PR.

tilman1, 2020-10-03:
Hi,
my fork should work for the ESP32. I tried it on a doit devkit v1 and an olimex ethernet. And it still compiles and works for the ESP8266. 
@dirkmueller 
From the way of how the firmware is written, i.e. processor variants by ifdef and determination of sensor configuration at runtime, I assume that the underlying paradigm is to have one source code that can cover as many platforms and sensors configuration (variants) as possible only limited by the memory available.
- What is the outlook for the ESP32 in the future? Is there the intention to bring to a release branch as well?
- I would like to merge the ethernet support for the olimex. At the same time, I would not want to violate any implicit design paradigms or (system) requirements. Is there an owner of the code or some sort of review process?

Thanks
Tilman

ro85ac, 2020-10-04:
Hello everyone,

I am new to arduino and esp32. I have a similar situation. I am receiving the same error: .../freertos/queue.c:1442 (xQueueGenericReceive)- assert failed!

I am not using the WiFi connection, but the GSM one. But I have no idea how to fix it.
As a note, using the wifi connection the messages are sent to the Azure Cloud.

[SIM7000G_Azure_MQTT.txt](https://github.com/opendata-stuttgart/sensors-software/files/5324288/SIM7000G_Azure_MQTT.txt)

primusmagestri, 2020-10-05:
@ro85ac see the comment above from @peterhinson , he added some fixes in the beta branch. I did sort of the same fix on my copy based on what he recommended and it worked.

btw I'm not sure I understand what you achieved with wifi and Azure since that runtime error breaks everything before you start configuring the device...

ro85ac, 2020-10-05:
> @ro85ac see the comment above from @peterhinson , he added some fixes in the beta branch. I did sort of the same fix on my copy based on what he recommended and it worked.
>
> btw I'm not sure I [...]

@primusmagestri  if connecting by wifi it works, while by gsm it throws error?

pjgueno, 2020-10-05:
Hi. This week end @hbitter made the firmware work n an Heltec ESP32. I made a pull request here for him but you can also check in my own GitHub: https://github.com/pjgueno/sensors-software/tree/beta
I put a lib he added in the "lib" folder if needed.

ro85ac, 2020-10-05:
@pjgueno  I don't see any connection with the libraries you've sent and the error I mentioned before. I am an amateur when it comes to Arduino and maybe I don't see the right library.

peterhinson, 2020-10-06:
@ro85ac - in my case, the `(xQueueGenericReceive)- assert failed!` error was related to calling `configTime()` before a TCP/IP connection was established (presumably Wifi or GSM). Take a look at [this commit](https://github.com/opendata-stuttgart/sensors-software/pull/796/commits/5959da3d320ca37b70419be5931dded5abb99025) for a workaround, adding `tcpip_adapter_init()` prior to calling `configTime()` seems to resolve it.