opendata-stuttgart/sensors-software #788: Gas sensor support?

jpsingleton, 2020-09-11:
Is there any appetite for supporting gas sensors? For example, the [MiCS-6814](https://www.sgxsensortech.com/content/uploads/2015/02/1143_Datasheet-MiCS-6814-rev-8.pdf) (CO/NH3/NO2, 5V, analogue, ~$20 USD).

I'm particularly interested in measuring NO/NO2/NOx in addition to PM and T/H/P. The Plume Flow measures NO2 and VOC in addition to PM and these readings are very useful, as the pollution sources are often the same (e.g. diesel combustion).

KoffeinKaio, 2020-09-15:
MiCS-6814 required 3 adc channels - esp8266 only has one - so you would need some sort of extender / i2c adapter board there.

ESP32 has the channels but the MiCS-6814 is 5V and ESP32 3.3 - so level shifter here.

For this specific sensor it would require too much work I think to be included, as I cant find a ready-i2c board for this.

OperKH, 2020-10-06:
MICS-6814 can be connected via ADS1115 via I2C, like it done in this firmware https://eco-city.org.ua/firmware/EcoTube-STABLE-1.0.8-0-ArduinoIDE.bin

ricki-z, 2020-10-18:
This firmware was / is developed for outdoor devices. The outdoor limit for NO2 is 40µg (or 21 ppb / 0.021 ppm). The lowest value the MICS-6814 can record is 50 ppb (95 µg).
We have to focus on sensors that are suitable for our use case. The memory of both the ESP8266 and the ESP32 is very limited. So we can't add all possible sensors only because they exist. Please check the data sheets of new sensors before asking us to include them.

jpsingleton, 2020-10-21:
It may not be suitable to ensure readings are below the 40 µg/m3 (annual mean) WHO/EU limit but it could show if this was breached significantly. It would also be useful for checking the 200 μg/m3 (1-hour mean) limit. Diffusion tubes are generally used for the annual measurements but these are of limited value as they are not real-time. However, as there appears to not be interest in supporting this I will close this and focus my efforts elsewhere.

closed 2020-10-21 by jpsingleton

ricki-z, 2020-10-21:
Even on streets with high traffic you will have values this high nowadays in most countries.
And there are other difficulties beside the lower limits of these sensors. We tried to implement NO2 measurements with another sensor type (Alphasense). Just try to implement this and look at the dependencies between temperature, humitidy and NO2 concentration. Try to find a mathematical correction for this with values that you can't compare to any official source or measurement equipment (because you won't find concentrations that often in "real life").
Other points are the selectivity of the sensor, the life time (for some of the cheaper sensors it's only 6 months) or the need of recalibration (who will do this, how should it be done). I couldn't find some of these informations in the datasheet.