opendata-stuttgart/sensors-software #789: Network instability with NRZ-2020-130-B8

Phaze-III, 2020-09-12:
After an OTA upgrade to NRZ-2020-130-B8 network connectivity became very unstable.

- increased number of WiFi errors (at least a few every hour, used to be practically none)
- increased number of upload errors, including to the local InfluxDB, resulting in a lot of measurements gaps
- WebUI unresponsive or slow (timeouts and empty responses) most of the time

Reverting back to the NRZ-2020-129 release firmware makes the sensor stable again. NRZ-2020-130-B6 also was very stable.

Phaze-III, 2020-09-12:
System parameters fetched every 15 minutes from status page for period upgradeing from 130-B6 to 130-B8:
[image]

System parameters for period downgrading from 130-B8 to 129-release:
[image]

dirkmueller, 2020-09-14:
@Phaze-III thanks for the statistics, that's very helpful. In B8 we switched to a newer version of core. However, the wifi code etc (so the SDK) remained the same version like before. So for now I would assume that there is some other problem that we can trigger elsewhere as well if we know how.

Can you describe which features you configured on your airrohr? Which sensors are enabled? Which APIs? Do you use any displays?

Is there anything noteworthy about your WiFi setup? What's the wifi quality (RSSI) you that you get?

Phaze-III, 2020-09-14:
Hi @dirkmueller, thanks for looking into this.

**Sensoren**
- SDS011 (Feinstaub)
- BMP280/BME280 (Temperatur, rel. Luftfeuchte, Luftdruck)

**APIs**
API Sensor.Community  (HTTPS)
API Madavi.de  (HTTPS)
Senden an InfluxDB  (**no** HTTPS)

No displays (the sensor is placed outside)

WiFi access point is a Cisco EPC 3928 (both 5Ghz and 2.4 Ghz, using different SSID's). The 2.4 Ghz band is rather busy (lot of neighbours) but still relatively good when using a fixed channel assignment. The AP is setup as N-only and the default 20Mhz channel width.

Quality as measured by the sensor is between 70% and 80% (-70dBm to -60dBm).

As for the symptoms this does remind me of the instability I had when connecting to the WebUI with NRZ-2019-125-B1 (if I remember correctly, a version before NRZ-2020-129). Since NRZ-2020-129 the sensor was very stable and no problems connecting to the WebUI.

Looking at the statistics from the last two days with a switch from 129 to 130-B8 and to 130-B9 early this morning, the 'Samples' graph shows significant differences between the versions. See included screenshot. Might be a clue somehow?

[image]

imaluschi, 2020-09-15:
Hi, I'm not really sure if my sensor-problems are the same issue, but it seems so.

Firmware NRZ-2020-130-B9/DE (Sep 13 2020)

Sensors
* SDS011 (registered)
* DHT
* BME280 (for comparing with DHT, especially humidity)
* DNMS (registered)

APIS:
API Sensor.Community (HTTP)
API Madavi.de (HTTP)
own Api (Server with php-script for saving DNMS-values to csv and MySQL database).

Today morning i realized, that the DNMS-Sensor did not send measurements during the night.
[image]

When checking the PM-values, there are sent measurements but with gaps.
[image]
WIFI-Signal shows also gaps.
[image]

But this is not an explanation for a full breakdown for the dnms. Furthermore the BME-values are crap. WIFI Signal Quality is obviously changing fast.

[image]
[image]

These crappy values occured since the instability in the night.
[image]

Temporary (unsatisfying) Solution: put the USB-cable out and in from the power supply - all values are ok again.
[image]

[image]

Will switch back to the latest stable release.

dirkmueller, 2020-09-15:
@imaluschi I think this sounds like a different issue. please file a new ticket and please let us know if the issue returns with beta firmware after power cycle. 

its possible that there is a specific quirk that we need to add for DNMS to survive an OTA.

dirkmueller, 2020-09-15:
> Looking at the statistics from the last two days with a switch from 129 to 130-B8 and to 130-B9 early this morning, the
> 'Samples' graph shows significant differences between the versions. See included screenshot. Might be a clue somehow?

@Phaze-III thanks for sharing that telemetry data, that is very useful. so from your description this sounds very similar to my setup, which works fine. The key difference is that I disabled HTTPS for sensor.community and madavi.de. can you confirm that -B9 works fine for you when you have HTTPS disabled in general?

The low "samples" that you show are the # of loop() entrances we had since last measurement. a fluctuating or low value here means that the sensor is busy on something that means the loops don't quit rapidly enough. with low loop() counts, wifi issues and other instabilities are a result, because all the housekeeping and wifi / network processing is only done outside the loop. 

so we know how the wifi issues are caused but we need to identify which part of the code causes the problem. I suspect HTTPS so please turn it off, powercycle your machine and let us know how it goes for B9. 

so far I blamed this on issues with the server (lack of ssl ticket resumption support meaning the poor small cpu needs to do excessive work on creating a new TLS session rather than reusing the previously established one), but maybe there is more that can be done. I'll look at that over the next few days.

Phaze-III, 2020-09-16:
I loaded B9 again and first disabled HTTPS. Because Sensor.Community went down (I think) around that time I also disabled the Sensor.Community API and let the sensor run for the night. Sorry to say that it didn't help :( Still Wifi and upload errors and an unresponsive WebUI. Statistics included below.

I also tried some other things which are marked in the graph with annotation markers:
```
2020-09-15T20:55:00Z Disable API HTTPS and load 130-B9 (#789)
2020-09-15T21:51:48Z Disable Sensor.Community API
2020-09-16T08:53:00Z Disable Madavi.de (#789)
2020-09-16T09:24:47Z No local NTP (#789)
2020-09-16T11:41:35Z Disable local influxDB (#789)
2020-09-16T13:20:03Z NRZ-2020-129 (#789)
```
Even running without any uploads didn't help (still a very unresponsive WebUI).

I'm now back to 129 and will try to test with another AP/sensor location later.

[image]

Phaze-III, 2020-09-16:
Moving the sensor inside and on another AP with less neighbours the same results (timeouts and empty responses).

pjgueno, 2020-09-16:
I can be the ESP12 chip on the ESP8266 itself. I saw it once with a custom PCB fron Air Gloucester : The ESP12 died for Wifi suddenly... It could not connect and I had to ask for a new PCB... Can you check the USB-logs

Phaze-III, 2020-09-16:
Wifi is working fine again when switching back to NRZ-2020-129 so I don't think it is the hardware. Any help with getting to the USB logs would be appreciated. Thanks

pjgueno, 2020-09-16:
If you are on MacOS:
`ls /dev/tty.*`
`screen /dev/tty.wchusbserialfd140 9600 ` with tty.wchusbserialfd140 = the ESP8266 (can difer a bit)
To stop:
`screen -ls`
`screen -X -S 2280.ttys001.MacBook` the number can difer

You should see the log of the firmware

You can also use the monitor in the Arduino IDE.

Phaze-III, 2020-09-16:
Thanks! This is what happens (I have only the local InfluxDB enabled):
```
... Many Start/End reading SDS011
Start reading SDS011
End reading SDS011
Start reading SDS011
End reading SDS011
Start reading SDS011
End reading SDS011
Start reading SDS011
End reading SDS011
Start reading SDS011
PM10 (sec.) : 17.30
PM2.5 (sec.): 11.60
End reading SDS011
Start reading SDS011
PM10 (sec.) : 16.80
PM2.5 (sec.): 11.30
End reading SDS011
Start reading SDS011
PM10 (sec.) : 17.00
PM2.5 (sec.): 11.30
End reading SDS011
Start reading SDS011
PM10 (sec.) : 16.60
PM2.5 (sec.): 11.20
End reading SDS011
Start reading SDS011
PM10 (sec.) : 16.90
PM2.5 (sec.): 11.30
End reading SDS011
Start reading SDS011
PM10 (sec.) : 16.40
PM2.5 (sec.): 11.20
PM10:  16.83
PM2.5: 11.27
----
End reading SDS011
Start reading BMP/E280
Temperature (°C): 24.02
Pressure (hPa): 102318.03
Humidity (%): 57.59
----
End reading BMP/E280
## Sending to custom influx db: 
Parse JSON for influx DB: {"software_version": "NRZ-2020-130-B9", "sensordatavalues":[{"value_type":"SDS_P1","value":"16.83"},{"value_type":"SDS_P2","value":"11.27"},{"value_type":"BME280_temperature","value":"24.02"},{"value_type":"BME280_pressure","value":"102318.03"},{"value_type":"BME280_humidity","value":"57.59"},{"value_type":"samples","value":"2154296"},{"value_type":"min_micro","value":"51"},{"value_type":"max_micro","value":"20294"},{"value_type":"signal","value":"-44"}]}
Succeeded - 10.0.2.2
Time for Sending (ms): 17
Start reading SDS011
End reading SDS011
Start reading SDS011
End reading SDS011
Start reading SDS011
End reading SDS011
Start reading SDS011
End reading SDS011
Start reading SDS011
End reading SDS011
Start reading SDS011
End reading SDS011
Start reading SDS011
End reading SDS011
Start reading SDS011
.....
```

pjgueno, 2020-09-16:
Can you send the first lines after reboot to seeif the sensor find the network ? 
It seems good. 
You have an influxDB instance?

Phaze-III, 2020-09-16:
I have a local InfluxDB instance. First lines after reboot:
```
airRohr: NRZ-2020-130-B9/DE
mounting FS...
opened config file...
parsed json...
output debug text to displays...
Connecting to Phaze-III-LS
....SNTP synced: Wed Sep 16 19:59:00 2020

twoStageOTAUpdate
Start reading SDS011 version date
End reading SDS011 version date
HTTP GET: firmware.sensor.community:443/airrohr/beta/latest_de.bin.md5
GET r: 200
output debug text to displays...
No newer version available.
.
WiFi connected, IP is: 172.24.2.90
Starting Webserver... 172.24.2.90

ChipId: *********

MAC Id: ***************
Read SDS...: 18-11-16(0bb3)
Stopping SDS011...
Read BMP280/BME280...
Trying BMP280/BME280 sensor on 76 ... found
Send to :
custom influx DB
----
```

So the network is found, uploads to the local influxdb most of the time succeeds but the WebUI is mostly not responding.

pjgueno, 2020-09-16:
You mean the WebUI through the local network? or devices.sensir.community?
I think it is some server problem on our side.

Phaze-III, 2020-09-16:
The webUI of the local sensor.

pjgueno, 2020-09-16:
Is your sensor not to far away from the AP ? Do you use 2.4 GHz and 5Hz networks with the same SSID and Password on the same channel (I already saw that...) ? I already see the case. Actually some of my 4 sensors at home may be sometimes difficult to load but I never succeed to understand and reproduce the case. What kind of router do you have ? an OEM one or someting else ? Here in Germany some Vodafone router won't recognize the ESP easily...

Phaze-III, 2020-09-16:
See https://github.com/opendata-stuttgart/sensors-software/issues/789#issuecomment-692310239 for the details of my setup. And I try with various methods and devices, ranging from curl to read the status page every 15 minutes to laptop, phone and tablet :)

pjgueno, 2020-09-17:
Actually, you had already wrote everything above. My fault... Is there a WPS and/or a guest network?

Phaze-III, 2020-09-17:
No worries, they were all valid questions. WPS is enabled on my AP but not used. It doesn't have a guest-network.

You also wrote:
> Actually some of my 4 sensors at home may be sometimes difficult to load but I never succeed to understand and reproduce the case.

I also noticed that with the stable firmware. That was one of the reasons I started to collect the statistics from the status page every 15 minutes. Most of the time it was related to periods with an increased "Dauer Messübertragung". For example when api.sensor.community or madavi.de was very busy and upload times peaked to 20 to 40 seconds. I guess when the sensor is uploading it can't serve webpages.

ituri, 2020-09-21:
This doesn't have anything to do with the original question, but: @Phaze-III, could you elaborate on how you get these nice statistics? They look like Grafana to me but how do you aggregate the data?

Phaze-III, 2020-09-21:
@ituri : please have a look at https://github.com/Phaze-III/luftdaten-sensor-status . That's more or less what I use to get the stats.

Phaze-III, 2020-09-21:
I've been doing a lot of testing with locally build images and I now have a well-performing 130-B9 image. 

Long story short: it looks like the code added to fix #781 somehow makes my sensor unstable. Going back to commit 8ebc874de5e8211fc2afa1069ca21cff18c685ab and reverting the code changes in commit 5ba7ed8101dad465c27b1c8c34008dbbfc2bc44e the resulting firmware runs smoothly with a "Samples" rate comparable with NRZ-2020-129. Most notably it has a very responsive WebUI. The ref for this build is at Phaze-III/sensors-software@dd2c958d45566af619bce8c554680df86679c313

Note that even the code change in 5ba7ed8101dad465c27b1c8c34008dbbfc2bc44e had a noticeable effect and I suspect this commit was also in the public NRZ-2020-130-B8 binary.

So the good news is that the update to Core 2.7.4/3 is not the cause. Why the GPS fix code has this effect on my (and perhaps other?) sensor is the question.

Just in case: my board is a RobotDyn WIFI NodeM ESP8266, CP2102 (32M flash) https://robotdyn.com/wifi-nodem-esp8266-cp2102.html

dirkmueller, 2020-09-21:
@Phaze-III so you do not have GPS enabled but you are logging into an influxdb?

dirkmueller, 2020-09-21:
I think the real struggle comes from https://github.com/opendata-stuttgart/sensors-software/commit/194beeeea679068d2cc5e2bb0d37118658585c25 

The code could be improved in several ways.

dirkmueller, 2020-09-21:
Can you try with that patch reverted only? To resolve conflicts maybe just do the influx string part, the rest of the patch can be kept.

Phaze-III, 2020-09-21:
@dirkmueller 
> so you do not have GPS enabled but you are logging into an influxdb?

Correct.

I reverted the isNumeric commit and that does indeed improve things. Still a bit 'slower' than the build without any of the GPS commits but much better than the online B9. 

The "Samples" graph:
[image]

Phaze-III, 2020-09-22:
The patched firmware has been running for a night and I noticed that there still are wifi and upload errors, in numbers comparable with the OTA upgrade to B8 which triggered this issue. So I also reverted part of commit 5ba7ed8101dad465c27b1c8c34008dbbfc2bc44e (by commenting out one code-block, see Phaze-III/sensors-software@b6801b296d12f0323018d7fa18f0477cd0877eee).

After running for 5+ hours now no wifi-errors yet (still some upload errors but those appear to be the normal glitches) and the number of Samples is at a steady rate of around 4500000 (same as NRZ-2020-129). See graphs below.

What I'm understanding from this is that code used to process values read from a GPS-module somehow seems to interfere with the performance of my sensor, even though I don't have one and haven't enabled one in the config. Should that code be reached at all?

**Sensor configuration**
[image]

**Fehler graph**
[image]

**Samples graph**
[image]

dirkmueller, 2020-09-23:
> even though I don't have one and haven't enabled one in the config. Should that code be reached at all?

Exactly thats what I'm scratching my head on. if the GPS sensor isn't enabled (as shown in your screenshot it isn't) the code you're touching isn't executed: 

from airrohr-firmware.ino around line 4400:

```c++
>	if (cfg::gps_read && !gps_init_failed) {
		// process serial GPS data..
		while (serialGPS->available() > 0) {
			gps.encode(serialGPS->read());
		}

		if ((msSince(starttime_GPS) > SAMPLETIME_GPS_MS) || send_now) {
			// getting GPS coordinates
>			fetchSensorGPS(result_GPS);
			starttime_GPS = act_milli;
		}
	}
```

the code you have modified is fetchSensorGPS, which is inside an if() conditional that should be always false. 

can you please try with just the isNumeric commit reverted? I somehow suspect the isNumeric string/value conversion is for some reason (that I do not understand) sometimes flipping the datatype from value to string and vs. influxdb is very sensitive to data type changes, and it will outright reject upload when the datatype doesn't match. 

I have started rewriting this code with my test sensor. Although I can not spot an error in the code that would explain all of this.

dirkmueller, 2020-09-23:
@Phaze-III another comment on this. the "concrete" absolute value of "Samples" doesn't matter that much. The only thing that matters is that it is higher than 1000 times the "measurement_interval in seconds", and that the numbers don't fluctuate too much. 

the real number can change between each firmware build, because the code runs from a builtin flash, that is comparably slow to the cpu (we are clocking the cpu at 140MHz, and depending on your particular model variant, your flash is clocked between 10 and 40Mhz). so every memory access to the flash makes the cpu wait huge amount of time (at the scale of the performance of the cpu). there is complicated alignment and caching effects involved, so just a recompile of the binary with marginal code changes here and there (that never get executed) change alignment of huge parts of the code, and change code runtime drastically. so please do not put too much emphasis on this part. 

the real interesting thing is whether the upload dropouts with your influxdb are disappearing, and if they disappear, if that fixes the "wifi issues".

dirkmueller, 2020-09-23:
I think there is for some reason a very strong correlation between the samples time and your wifi instability. could you please post output of "esptool.py flash_id" (potentially you need to specify the path to the device in addition. 

the only other thing that we could be doing is giving the compiler better hints at how to partition and align the code so that the "hot" part of the loop runs quickly.

Phaze-III, 2020-09-23:
@dirkmueller Thanks for the explanation on the interpretation of the "Samples" and the changes between firmware builds. That explains why I had a good performance with B6 even though the "Samples" rate was lower.

> could you please post output of "esptool.py flash_id"
```
$ esptool --port /dev/ttyUSB0 flash_id
esptool.py v2.8
Serial port /dev/ttyUSB0
Connecting....
Detecting chip type... ESP8266
Chip is ESP8266EX
Features: WiFi
Crystal is 26MHz
MAC: dc:4f:22:60:72:dc
Enabling default SPI flash mode...
Manufacturer: ef
Device: 4016
Detected flash size: 4MB
Hard resetting via RTS pin...
```

> can you please try with just the isNumeric commit reverted?

Results of a build with **only** the isNumeric commit reverted did improve things but still with wifi-errors (see my previous comment). I now re-enabled the isNumeric commit and only left the out-commented code in fetchSensorGPS(). That build seems to be very stable after running for an hour now. This suggests that the main trigger for the wifi instability seems to be the fetchSensorGPS() change even though it doesn't get executed so I'm now also thinking that this might be an alignment issue.

> the only other thing that we could be doing is giving the compiler better hints at how to partition and align the code so that the "hot" part of the loop runs quickly.

That would be unknown territory for me but I'll be happy to test.

imaluschi, 2020-10-03:
Tried once again to Update to the last beta NRZ-2020-130-B9 (via Auto-Firmware-Upgrade).
Missing values since then, especially DHT and BME.
[image]
[image]
[image]

With a newly installed WLAN-Repeater the signal quality is around 95-100%, so the WLAN supply is not the problem (as it maybe was also a problem earlier).
Switching back to latest stable release (NRZ-2020-129) which is working fine.

Phaze-III, 2020-10-03:
As a test I started to build the firmware using the Arduino IDE instead of PlatformIO. I used a clean checkout of the current beta branch and set Arduino settings to get the same build options as with the current platformio.ini. Additional libraries used are the exact same ones as found in .pio/libdeps. The resulting firmware is slightly smaller and runs very stable. 

The settings for Arduino IDE (including some additional defines in platform.local.txt):
[image]
```
$ cat Library/Arduino15/packages/esp8266/hardware/esp8266/2.7.4/platform.local.txt
build.extra_flags=-DESP8266 -DHTTPCLIENT_1_1_COMPATIBLE=0 -DNO_GLOBAL_SERIAL=0 -DINTL_DE
```
So somehow the Arduino IDE builds 'better' binaries than PlatformIO.

dirkmueller, 2020-10-07:
So I just got back to this, and looked a bit at the changes that were pointed out as problematic here. What we noticed is that the addition of "not executed code" caused some issues to appear. I noticed that with the addition of the GPS timestamp handling, as this function got inlined by the compiler, the loop() function for the first time got bigger than 8192 bytes (8225 bytes in my build). This is the only major change I could see by comparing the two resulting firmware binaries. 

So I went ahead and optimized the code size down of loop() back to around 6900 bytes, and it doubled the "samples" counter for me. it for some reason still isn't as good as with the patches reverted (which makes little sense so far to me), but its definitely better. 

A build can be downloaded from here: 

https://static.dmllr.de/airrohr/beta/builds-NRZ-2020-130-B10/

I have two near-identical test nodes, and on one I now have about 1.5 million loops executed, and on another around 1.1 million. How there can be a 40% difference on an identical binary with compareable settings I so far do not understand. 

I would like to get your feedback if the build linked on the website above improves the situation for you or not.

ricki-z, 2020-10-07:
I have uploaded this version as the actual beta. So we can have a look on the devices using the beta version.

Phaze-III, 2020-10-08:
@dirkmueller I flashed the firmware from dmllr.de and had it running for the night. That build runs very stable, no Wifi errors, very responsive UI. Samples stable at 3.6M. 

The bad news is however that the actual OTA build is still very unstable. I did an OTA downgrade to 129 and then let the sensor do an OTA upgrade to B10. Statistics below.

The firmware I downloaded from dmllr.de has a different size than the B10 on firmware.sensor.community:

https://static.dmllr.de/airrohr/beta/builds-NRZ-2020-130-B10/latest_de.bin
548832 Oct  7 22:33 latest_de.bin

https://firmware.sensor.community/airrohr/beta/latest_de.bin
548768 Oct  8 01:42 latest_de.bin

I know that there can be small differences in size or in actual bytes but in this case I would have expected that both binaries would be the same (except perhaps for 1 byte for the build date).

A local build with the current beta branch gives the same binary as on firmware.sensor.community.

I'm not quite sure what to make of this but what you did to get the build on dmllr.de seems different than what building from the beta branch now gives.

[image]

dirkmueller, 2020-10-08:
@Phaze-III thanks for the testing, really appreciate helping  us out here.

Can you please manually flash the sensor.community build and see if that changes things? Please power it off beforehand for 2-3 min.

I begin to suspect that there is some effect in play when using OTA. The ota upgrade process is fairly complicated, and it isn't unheard of problems in that area. We might not actually have the build on flash that you have downloaded.

I will now search for differences in those two firmware files

dirkmueller, 2020-10-08:
It is good that you confirm that there's not an issue with three build that I provided, because I see no issues here either 🤷

Phaze-III, 2020-10-08:
Manually flashing the sensor.community build didn't help, same results and instability as after the OTA upgrade. The OTA upgrade I did was OK I think, after the upgrade it did mention the Oct 8 2020 build date (your build has Oct 7 2020).

I'm leaning towards differences in building environment/process.

Phaze-III, 2020-10-08:
Just remembered I have a screenlog of the OTA update to B10, looks like the correct firmware was found and loaded. It did have startup problems (failed to NTP sync soon enough) but the actual Dauertest was done after a power off/on.
```
airRohr: NRZ-2020-129/DE
mounting FS...
opened config file...
parsed json...
output debug text to displays...
Connecting to <REDACTED>
...........
WiFi connected, IP is: 10.0.2.90
Starting Webserver... 10.0.2.90

ChipId: 6320860
Read SDS...: 18-11-16(0bb3)
Stopping SDS011...
Read BMP280/BME280...
Trying BMP280/BME280 sensor on 76 ... found
Send to :
sensor.community
Madavi.de
custom influx DB
----
Auto-Update active...
SNTP synced: Thu Oct  8 09:36:32 2020

twoStageOTAUpdate
Update md5: 10fd9680bfbe16c86a4c7cdf21851fde
Sketch md5: f6f8635ad2b0f0efe43f70a14a6ee3f9
Success downloading: /airrohr/beta/latest_de.bin
Success downloading: /airrohr/beta/latest_de.bin.md5
Success downloading: /airrohr/loader-002.bin
launching 2nd stage
Erasing SDK config.
Restart.
?!L?<1?1P?Found firmware MD5: 10fd9680bfbe16c86a4c7cdf21851fde

Moving Firmware image to old.
Erasing SDK config.
Finished successfully.. Rebooting!
?!L?<1?1P?airRohr: NRZ-2020-130-B10/DE
mounting FS...
opened config file...
parsed json...
Rewriting old config from: NRZ-2020-129
Saving config...
Config written successfully.
output debug text to displays...
Connecting to <REDACTED>
.........
WiFi connected, IP is: 10.0.2.90
Starting Webserver... 10.0.2.90

ChipId: 6320860

MAC Id: dc4f226072dc
Read SDS...: 18-11-16(0bb3)
Stopping SDS011...
Read BMP280/BME280...
Trying BMP280/BME280 sensor on 76 ... found
Send to :
sensor.community
Madavi.de
custom influx DB
----
Auto-Update active...
PM10:  3.97
PM2.5: 2.70
----
## Sending to sensor.community - SDS011
Time incorrect; Disabling CA verification.
Succeeded - api.sensor.community
Temperature (°C): 15.31
Pressure (hPa): 100957.66
Humidity (%): 91.02
----
## Sending to sensor.community - BMP/E280
Time incorrect; Disabling CA verification.
Succeeded - api.sensor.community
## Sending to madavi.de: 
Time incorrect; Disabling CA verification.
Succeeded - api-rrd.madavi.de
## Sending to custom influx db: 
Succeeded - 10.0.2.2
Time for Sending (ms): 2115
ws: root ...
ws: values ...
ws: status ...
SNTP synced: Thu Oct  8 09:42:32 2020

twoStageOTAUpdate
output debug text to displays...
```

dirkmueller, 2020-10-09:
@Phaze-III okay, can you please try the firmware  from https://static.dmllr.de/airrohr/beta/builds-NRZ-2020-130-B10-fp-in-rom/ ?

It isn't clear anymore why I had produced a firmware with a different size than the official builds, it appears like I had local patches to my arduino core checkout. a wipe of the checkout (unfortunately I failed to properly back it up before hand) reproduced the official firmware binary size. content was still  different. it looks like every build has a very different code layout due to individual functions being ordered differently in the binary. 

so I hardcoded the -frandom-seed to a fixed value, and I changed the Flaoting point code to move to irom again instead of iram, which frees up memory a bit. This produced a much higher "samples" value for me.

of course thats just a temporary hack. since I am not changing the source code at any point in time, just the way the binary is being linked together, I still need to find out what the issue is. it starts to smell like padding and alignment settings that are making large differences.

Phaze-III, 2020-10-10:
@dirkmueller This build has been running for 11 hours now and is extremely stable. Samples 5M+ and stable, no Wifi errors and an always very responsive UI 👍 
[image]

dirkmueller, 2020-10-10:
@Phaze-III Okay, great. so while I can't reproduce the exact issue you're reporting, it seems strongly correlated with low samples counters, so this is something that I can watch out for and analyze further. for now I consider this problem solved well enough (I hope Rajko creates a B11 with the change), so that we can finally do a stable release again, after over a year :-)

ricki-z, 2020-10-11:
B11 is online as the latest beta firmware.

ricki-z, 2020-10-11:
@dirkmueller latest release version NRZ-2020-129 is online since 2020-01-07 😉 .

dirkmueller, 2020-10-11:
@ricki-z oh, indeed! Then is only 9 months..

Phaze-III, 2020-10-11:
@dirkmueller 
> for now I consider this problem solved well enough

Thanks for this solution! I also did the OTA-down/upgrade dance and everything runs fine. I hope this also helps other people :)

ricki-z, 2020-10-13:
@Phaze-III can you give us a short feedback if it's still okay after some days running?

dirkmueller, 2020-10-13:
the random-seed might make the build between machines more reproducible, not sure. I think it shouldn't harm for now. we can eventually drop this once the *actual* problem has been understood. 

the observation of using B9 with -DFP_IN_IROM also running stable for you is very relevant. this would mean, if reproducible, that the samples counter optimisation was a false red herring (to some extend that is a relief for me). This setting only changes how floating point operations are handled. we're falling back to the in-rom floating point code rather than the one provided by the C compiler. 

I am not seeing "Exception" as a reset reason in your telemetry, which should imho happen if the node crashes on floating point exceptions. However, changing the floating point handlers for functions changes the behavior. that is very odd. I begin to wonder if we have an issue with handling certain values in some of the web sites somewhere. 

which urls on the device are you regularly fetching for collecting all these telemetrics that you screenshotted? maybe that is related to the issue we have.

Phaze-III, 2020-10-13:
I haven't seen Exception-resets with the -DFP_IN_IROM builds. The last Exception I had was with the NRZ-2020-130-B10/DE (Oct 7 2020) OTA build which was still unstable (before the -DFP_in_IROM) so that definitely could have been caused by floating point exceptions. 

The telemetry comes from two sources:
1. The system data sent to the local influxdb by the sensor (max_micro min_micro samples signal) every Messintervall
2. Parameters as reported on de /status page fetched every 15 minutes from the sensor and stored in a separate influxdb

So basically all useful info you've put in there :-)

The script to fetch and 'parse' the /status page and Grafana dashboard to display the data is available on https://github.com/Phaze-III/luftdaten-sensor-status

dirkmueller, 2020-10-19:
Closing this one as fixed, continuing the saga in the followup #814

closed 2020-10-19 by dirkmueller