opendata-stuttgart/sensors-software #790: BMP280 data does not show on luftdaten map, PMS5003 data does

pe0ter, 2020-09-13:
A few days ago I build my esp8266-3018027 airsensor, containing a PMS5003 and BMP280. Everything seemed to work as per instructions and all measurement data is visible on https://api-rrd.madavi.de/grafana/d/q87EBfWGk/temperature-humidity-pressure?orgId=1&var-chipID=esp8266-3018027 but the BMP280 temp/humidity/pressure do not show on the https://maps.sensor.community/ map with the default BMP280 PIN = 3, if I change the PIN to 11 (BMP180), Temperature and Pressure data does show on the map so it seems my data cannot be found proper with the default PIN = 3.
Does anyone know what is wrong?
Best regards,
Peter

ricki-z, 2020-09-14:
If you look at the legend of the graph you will see that you have a BME280. The difference between the BME280 and the BMP280 is not only a letter. The BMP280 can measure temperature and pressure (just like the BMP180) while the BME280 is also measuring humitidy. That's also why BMP180 and BMP280 are using the same API pin.

pe0ter, 2020-09-14:
Thank you @ricki-z, changing my sensor to did the trick.

ricki-z, 2020-09-14:
@pe0ter so we can close this issue?

pe0ter, 2020-09-14:
Sorry @ricki-z, yes we can close this issue and sorry for asking this, I should have seen it myself in de legend as you mentioned. I am preparing to also build a noise sensor and will double and triple check before asking questions on that one.

closed 2020-09-14 by ricki-z