opendata-stuttgart/sensors-software #793: Telemetry is okay, yet the measurement falters

Vendrel, 2020-09-14:
Hi! We have 35 devices in our administration and we experience the same glitch on several ones, time to time.
https://api-rrd.madavi.de/grafana/d/GUaL5aZMz/pm-sensors?orgId=1&var-chipID=esp8266-13753647&theme=light
https://api-rrd.madavi.de/grafana/d/GUaL5aZMz/pm-sensors?orgId=1&var-chipID=esp8266-13753655&theme=light
https://api-rrd.madavi.de/grafana/d/GUaL5aZMz/pm-sensors?orgId=1&var-chipID=esp8266-13753396&theme=light

If you compare their measurement charts with their wifi signal chart then you can see that there was no reason in the wifi signal for the falters in measurement submissions and also, the devices could submit the wifi signal continuously even when the measurement data submission omitted. 

For example:
Measurements: [image]

Telemetry: [image]

Device restart always resolves the problem, for a while. :)

dirkmueller, 2020-09-15:
so I look at https://api-rrd.madavi.de/grafana/d/Fk6mw1WGz/wifi-signal?orgId=1&var-chipID=esp8266-13753396&from=1600150301462&to=1600164410063 and I can see corresponding dropouts in wifi reporting and in pm sensor reporting. 

can you please share a screenshot of the /status page when this happens?

Vendrel, 2020-09-16:
This should be a simple wifi issue if there's a correspondence. Anyway, of course, I'll bring a screenshot right at the next occurrence. :)

Vendrel, 2020-10-10:
I happened to catch another appearance of the glitch WITH successful request of the screenshots from the user (just didn't find the thread here, for a while). (There was several appearance of the glitch but I could not get that screenshot from a layman user.)

And **you can see this actually happening** on the same device:
- Here: no measurement data for weeks: https://api-rrd.madavi.de/grafana/d/GUaL5aZMz/pm-sensors?orgId=1&theme=light&var-chipID=esp8266-14033804&vanid=igen
- Yet, wifi signal arrives. Weak though, but arrives: https://api-rrd.madavi.de/grafana/d/Fk6mw1WGz/wifi-signal?orgId=1&var-chipID=esp8266-14033804&theme=light

[image]

ricki-z, 2020-10-11:
The temperature/humidity values are send.
https://api-rrd.madavi.de/grafana/d/q87EBfWGk/temperature-humidity-pressure?orgId=1&var-chipID=esp8266-14033804
So it looks like a connection problem between the NodeMCU and the SDS011 (in this case).

Vendrel, 2020-10-11:
I see. Then this is a hardware issue or a software issue? May we hope that it could be fixed with a software update?

ricki-z, 2020-10-11:
This looks more like a hardware issue. Sorry.