opendata-stuttgart/sensors-software #794: Next PM Sensor enablement

dirkmueller, 2020-09-15:
The code for the Tera Next PM Sensor was merged but is currently disabled from compilation as it changes the baud rate for the serialSDS unconditionally. 

we need to make setup of serialSDS a post-configuration-reading initialization step

Also the code is quite large (more than 2kb) and contains a lot of copy&paste. needs to be refactored.

sumerland, 2020-09-23:
Is it "safe" to run the NextPM instead of the SDS (remove SDS)? Will the NextPM run continuously or is it going to sleep as the SDS? In case of the latter, is it ensured that after a wake-up to and prior to taking the measurement the heating is active long enough to be effective?

pjgueno, 2020-09-23:
I must proff read the code but it was already 6 months ago and I have never get any answer from Tera Group ever since. I am quite sure that I implemeted the kind of verifcation with an 8bit number. It should tell that the heating has heated well...

sumerland, 2020-09-24:
I was asking because the spec mentions a warm-up time of 10s.

sumerland, 2020-11-11:
I got an additional nodemcu with just the NPM connected to play with. I also fetched yesterday's beta branch. Could you please point me to the right places in order to enable the NPM code?

I understand that currently the NPM cannot run together with an SDS. Would it be OK to replace the SDS with the NPM on my current board (with SHT3X, SPS30 and BME280)?

pjgueno, 2020-11-11:
Maybe you should come to my own repo: https://github.com/pjgueno/AirRohrFirmwareNextPM/

This part is important:
`#if defined(ESP8266) && !NPM_READ
	serialSDS.begin(9600, SWSERIAL_8N1, PM_SERIAL_RX, PM_SERIAL_TX);
	serialSDS.enableIntTx(true);
#endif
#if defined(ESP8266) && NPM_READ
	serialSDS.begin(115200, SWSERIAL_8E1, PM_SERIAL_RX, PM_SERIAL_TX);
	serialSDS.enableIntTx(true);
#endif
#if defined(ESP32) && !NPM_READ
	serialSDS.begin(9600, SERIAL_8N1, PM_SERIAL_RX, PM_SERIAL_TX);
#endif
#if defined(ESP32) && NPM_READ
	serialSDS.begin(115200, SERIAL_8E1, PM_SERIAL_RX, PM_SERIAL_TX);
	Debug.println("SERIAL_8E1");
	pinMode(PIN_CS, OUTPUT);
	digitalWrite(PIN_CS, LOW);
#endif`

The Next pM works only with these options : 115200, SERIAL_8E1

You will have to configure each (software)serial the right way.

Then you have to add the support for the fetchsensorNPM function like all the other fetchsensor functions :

`  if (cfg::npm_read) {
    fetchSensorNPM(result_NPM);
  }`

The SPS30 works in I2C so there should be no problem. The other sensors are on other serials if I remeber well.

sumerland, 2020-11-11:
Thanks! The first block is already present in the current beta. I added 
`if (cfg::npm_read) { fetchSensorNPM(result_NPM); }`
but so far the sensor does not show up in the debug info... Will try again.

Do I have to connect pin 5 (CS)? And if so, to which pin on the nodemcu? I connected GND, 5V, TX and RX in the same manner as the SDS. The fan is running all the time, btw. No strong smell yet :-)

pjgueno, 2020-11-11:
Do I have to connect pin 5 (CS)? => No, it is useless in the last version of the internal firmware of the NextPM (I was told by someone at Tera).
Are you on a ESP32 or on a ESP8266?

sumerland, 2020-11-11:
ESP8266. The problem right now is that as soon as NPM_READ is 1 the GUI does not respond anymore.

pjgueno, 2020-11-11:
I need a bit of time to check my code. I tell you soon

sumerland, 2020-11-11:
I got a compile error using your repo... Something with PIN_CS undeclared

pjgueno, 2020-11-12:
Comment out #define PIN_CS D13 in ext_def.h

Modify my last commit so :

#if defined(ESP32) && defined(NPM_READ)
//REVOIR CONDITION POUR DEVKIT && NPM_READ
  serialSDS.begin(115200, SERIAL_8E1, PM_SERIAL_RX, PM_SERIAL_TX);
  Debug.println("SERIAL_8E1");
   // pinMode(PIN_CS, OUTPUT);
 // digitalWrite(PIN_CS, LOW);
#endif

And

      if(memcmp(data,answer_sleep,4)==0){
        serialSDS.flush();
        is_NPM_running = false;
        debug_outln_info(F("Next PM Sleep..."));
       }

	//else if

	if(memcmp(data,answer_power,4)==0){
        serialSDS.flush();
        is_NPM_running = true;
        debug_outln_info(F("Next PM Power On..."));
       }
	
	//else{
        //digitalWrite(PIN_CS, HIGH);
        //delay(500);
        //digitalWrite(PIN_CS, LOW);
        //is_NPM_running = true;
        //debug_outln_info(F("Next PM Reboot..."));
        //}

       break;

sumerland, 2020-11-12:
I assume I should apply those changes to your repo? The sensor ran fine for a few minutes but then I could not connect to the GUI anymore and the fan was running continuously.

sumerland, 2020-11-15:
Today I started from a fresh clone of the current beta branch and did the following changes:
1) add the checkbox for the NPM in the GUI
2) in ext_def.h NPM_READ to 1 (I guess that otherwise some of the NPM code will not be compiled)
3) added  the call of fetchSensor_NPM and the missing lines around data += result_NPM
4) changed the way the NPM results are treated in line with pjgueno's repo (otherwise the numbers were way too large)

It does work now but what I realized is that when the sensor is unplugged (maybe it depends at which stage) and powered up again the NPM is not initialized properly. The internal fan spins up on power-on but then enters a 2s cycle with a brief spin-up. I needed to flash again (including data wipe) to reset it properly. Together with one of my crappy BME280 (humidity way too low) it sits outside and values are more more meaningful than from SDS and SPS:

[image]

On the contrary

[image]

I wish I could see the NPM's data in api-rrd.madavi.de, too.

sumerland, 2020-11-15:
I was too quick... Values disappeared after approx 1h. Apart from that it seems to run OK (fan spin up/down as expected, webserver is responsive). Will try again tomorrow.

sumerland, 2020-11-18:
After some more testing with the abovementioned changes applied to firmwares 132-B3 and 131: 

- 132-B3: initial measurements looking good but GUI becomes unresponsive after a few minutes
- 131: runs fine for a couple of hours (about 4), then stops (fa
- at least for the 131 after powering off/on the board runs again & BME280 is reporting data but no values from the NPM

The only way to get it working again is to reflash & wipe data.

I am at a loss here but am happy to support anytime.

pjgueno, 2020-11-18:
Thanks for your support. 
I think there was a commit mixing...
"add the checkbox for the NPM in the GUI" => there was everything in my version. And I let the esp32 /NextPM work for a week without Problem.

I wish I had more time now to start over from my own code. I tried to look at my commits this weekend but I can't remember anything. As I told, the code was done in March/April and a lot was changed since then.

sumerland, 2020-11-18:
I have just flashed your version (based on 130-B5) again, including the changes you recommended a week ago. After a couple of successful measurements the values are missing now. The fan however is spinning up every 2 minutes. Any recommendation how to debug this? Is following the output in the debug level window in the GUI the only way?

sumerland, 2020-11-22:
Yesterday morning was quite cold with high humidity such that SPS30 and even more the SDS11 went through the roof. I was able to get some hours of data from the NPM and was surprised to see that the PM values were quite similar to the SDS & SPS. While the latter peaked at 30...40μg/m^3 the NPM showed about 27μg/m^3. 2h later with the sun out all three agreed on 5μg/m^3. I wonder if the heating is really active in the current versions.

pjgueno, 2021-11-19:
@sumerland Are you still interested in the Next PM. I made a new branch: beta-nextpm.
Would you like to check ?
I will remove the copy/paste stuff.
I am testing for the past 7 days.
There are still some points were the NextPM did not stop after reading...

sumerland, 2021-11-20:
Cool, I will give it a try and let you know how it works.

sumerland, 2021-11-20:
Apologies - for whatever reasons I cannot connect the nodemcu flashed with this firmware to my network. As I have seen issues with my Asus AC86U routers before, I added a TP-Link repeater just for the IoT devices. They all work - except this beast here. I even tried my old Asus N66, which also used to work well with IoT stuff and again the nodemcu won't connect. SSIDs are discovered, password is entered but after a short while the nodemcu is running as AP again.

I flashed the latest NAM firmware for testing purposes and it works as expected. Hence I believe the nodemcu is fine.

pjgueno, 2021-11-20:
I think I have found:

I put the init_config first to be able to change the Serial Name conditionnally => SDS pr NPM:

`init_config();

	if (cfg::npm_read)
	{
#if defined(ESP8266)
		serialNPM.begin(115200, SWSERIAL_8E1, NPM_SERIAL_RX, NPM_SERIAL_TX);
		serialNPM.enableIntTx(false);
		//(115200, SWSERIAL_8E1, NPM_SERIAL_RX, NPM_SERIAL_TX, false, 16, 160)????
		//In a more generalized scenario, calculate the bits (use message size in octets times 10) that may be asynchronously received to determine the value for isrBufCapacity in the constructor. Also use the number of received octets that must be buffered for reading as the value of bufCapacity. The more frequently your code calls write or read functions, the greater the chances are that you can reduce the isrBufCapacity footprint without losing data, and each time you call read to fetch from the octet buffer, you reduce the need for space there.
#endif
#if defined(ESP32)
		serialNPM.begin(115200, SERIAL_8E1, NPM_SERIAL_RX, NPM_SERIAL_TX);
#endif
		Debug.println("Read Next PM... serialNPM 115200 8E1");
		serialNPM.setTimeout(400);
	}
	else
	{
#if defined(ESP8266)
		serialSDS.begin(9600, SWSERIAL_8N1, PM_SERIAL_RX, PM_SERIAL_TX);
		serialSDS.enableIntTx(true);
#endif

#if defined(ESP32)
		serialSDS.begin(9600, SERIAL_8N1, PM_SERIAL_RX, PM_SERIAL_TX);
#endif
		Debug.println("No Next PM... serialSDS 9600 8N1");
		serialSDS.setTimeout((4 * 12 * 1000) / 9600);
	}
`

But this part come afterwards and breaks everything:

`void initNonTrivials(const char* id) {
		strcpy(cfg::current_lang, CURRENT_LANG);
		strcpy_P(www_username, WWW_USERNAME);
		strcpy_P(www_password, WWW_PASSWORD);
		strcpy_P(wlanssid, WLANSSID);
		strcpy_P(wlanpwd, WLANPWD);
		strcpy_P(host_custom, HOST_CUSTOM);
		strcpy_P(url_custom, URL_CUSTOM);
		strcpy_P(host_influx, HOST_INFLUX);
		strcpy_P(url_influx, URL_INFLUX);
		strcpy_P(measurement_name_influx, MEASUREMENT_NAME_INFLUX);

		if (!*fs_ssid) {
			strcpy(fs_ssid, SSID_BASENAME);
			strcat(fs_ssid, id);
		}
	}`

I will check if I can bring it up.

str3yt, 2022-01-13:
Just ordered a Next PM sensor and have various ESP8266 here... Willing to test!
@pjgueno I couldn't find the branch "beta-nextpm" anywhere on github. How can I download it?
@sumerland Have you been successful with your tests?

pjgueno, 2022-01-14:
https://github.com/opendata-stuttgart/sensors-software/tree/beta-nextpm/airrohr-firmware

pjgueno, 2022-01-24:
Merged in the beta firmware. It works:I am currently running a sensor with it : 67030 in Saarbrücken, Germany
[https://maps.sensor.community/#15/49.2380/6.9969](https://maps.sensor.community/#15/49.2380/6.9969)

ricki-z, 2022-01-26:
Could you please check where you pushed it to? I can't see a pull request for this.

str3yt, 2022-01-27:
Code has been merged into beta:
https://github.com/opendata-stuttgart/sensors-software/commits/beta

Still need some days until I can test (currenty waiting for delivery of connector so I can hoop up the Tera NextPM)

sumerland, 2022-02-01:
Honestly, I don't see any difference between the heated SDS011 in the NAM and the "cold" sensor in the SC case. I am going to post in the Forum about this soon. I also checked if PM2.5 makes more sense if I subtract PM1 but that looked completely wrong.

I will follow your data. Would it be possible to share the madavi links? That would make a comparison a bit easier.

What I haven't said: Many thanks for adding support for the NPM! It came quite unexpected but an excellent start into 2022 :-)

pjgueno, 2022-02-01:
As I have already said, it was a long and difficult process... But I am really happy that someone uses it!
I am doing the Piera Systems IPS-7100 now...

**2492:**
[https://api-rrd.madavi.de/grafana/d/GUaL5aZMz/pm-sensors?orgId=1&var-chipID=esp8266-1497630](https://api-rrd.madavi.de/grafana/d/GUaL5aZMz/pm-sensors?orgId=1&var-chipID=esp8266-1497630)

**70056**
[https://api-rrd.madavi.de/grafana/d/GUaL5aZMz/pm-sensors?orgId=1&var-chipID=esp8266-477983](https://api-rrd.madavi.de/grafana/d/GUaL5aZMz/pm-sensors?orgId=1&var-chipID=esp8266-477983)

pjgueno, 2022-02-02:
If you install your sensor with the Next PM for long term on the SC map, send me the location. I will forward the link to the people at Tera.

pjgueno, 2022-02-02:
I received some new from Tera today:
The AQMD in California has tested the Next PM:
[http://www.aqmd.gov/aq-spec/evaluations/summary-pm](http://www.aqmd.gov/aq-spec/evaluations/summary-pm)

[image]
[image]

They are quite happy at Tera.
I was sent the draft for the summary report. I should come soon o the AQMD page.

sumerland, 2022-02-02:
What does that mean? That in the field the PM1 value is actually representing PM2.1 to 4.6? That would match my observation (NPM PM1 in line with PM2.5 data from the other sensors).

I will put the NPM into a proper casing and place it next to the other sensors. I will then send a link for the folks at Tera.

Is the heating definitely activated in the current beta? Compared to the other sensors I saw much higher PM values last night, when humidity was high.

pjgueno, 2022-02-04:
I have got an answer from Tera:

Tee heating starts at once on Power On but the CTO of Tera tald me that it won't really be able to compensate efficiently the RH with the PowerOn/Sleep.

I should definitely do a 100 % PowerOn version.

sumerland, 2022-02-04:
Sounds good! 

Btw, my NPM has changed its behaviour throughout the past 24h. PM1 and PM2.5 have dropped. PM2.5 is now in line with the other sensors. Here is a 48h plot

[image]

str3yt, 2022-02-05:
Compiled and flashed the firmware using PlatformIO and VScode, I downloaded the "beta" branch code on January 30th from github, so commit 288365ff04388438f8df6e89498c35a452bd011f (from Jan 26th). Board is a NodeMCU V3. Once confronted with a new WiFi, the config page shows up and works fine! Even if I wait for a few minutes and reload from time to time, it is responsive all the time. Once I enter WiFi credentials and reboot, the board starts acting up. I can ping the IP and see it in the router, but if I enter it in a browser to access the config page again, the browser tab SOMETIMES shows the loading symbol forever... Yesterday evening it still worked more often and I could access the "current values" page and "debug" from time to time, but in between I got a lot of timeouts - very unstable. Yesterday it also uploaded some Wifi values to madavi, but after some minutes it stopped doing even that:
https://api-rrd.madavi.de/grafana/d/Fk6mw1WGz/wifi-signal?var-chipID=esp8266-6489612&var-type=system&var-query0=sensors&orgId=1

What I tried for debugging:
Tried different powersupplies and different Wifis
Tried running it for a few minutes with disconnected sensors, this made everything even worse (web interface not accessible at all)
--> nothing

I managed to access the config web interface and disabled the BME280, disconnected it and, as suggested by @pjgueno, set the debug level to 5
--> first values PM could be measured!

Of couse I was stupid and changed two things at a time...
I will monitor it for some time and then reconnect + reconfigure the BME280 (maybe it is a broken one, I will keep you updated)
The only strange thing I'm noticing at the moment: In "debug" I get the following message very frequently, every 2 seconds or so:
R/ Tera Sensor Next PM
/R Tera Sensor Next PM

...even when the sensor is not spinning at that moment.
Thanks for the great support! 😀

pjgueno, 2022-02-06:
I made a npm fulltime in the beta-piera branch but not tested.

pjgueno, 2022-02-06:
@str3yt 

R/ Tera Sensor Next PM
/R Tera Sensor Next PM

is in fetchsensorNPM() first and last lines.

It is just to check which function is played even between measurements.

Comment out.

str3yt, 2022-02-07:
One more comment about something that makes me wonder at the moment...
It seems that every few measurements, the NPM is spitting out exactly the same value for PM1, PM2.5 and PM10:
[image]
Also @sumerland posted a screenshot in this thread in November, showing the same values for all three measurements: https://github.com/opendata-stuttgart/sensors-software/issues/794#issuecomment-727641269
I've noticed that phenomenon quite often now... I wonder if it's an issue related to the serial communication since it seems very unlikely to me, that the actual measurement value is the same for all three particulate matter size bins!

Sensor is still running indoors at the moment for testing purposes by the way.

sumerland, 2022-02-07:
I've noticed that, too. And the uploaded data is very(!) noisy.

pjgueno, 2022-02-07:
I can't understand. I corrected that months ago. I have just checken: the word() byte mixing are ok and the variable are good. My sensor works well. Are you sure of your Version? Look Lines 3713 to 3719 and 3791 to 3797 and 1944 to 1949. Are the values the same in Serial?

str3yt, 2022-02-07:
I checked the source code files which I compiled, it is the same code as in the current beta branch here:
https://github.com/opendata-stuttgart/sensors-software/blob/288365ff04388438f8df6e89498c35a452bd011f/airrohr-firmware/airrohr-firmware.ino

My next step would be to test the sensor without the esp8266 (nodeMCU) but with directly connecting via one of the USB to RS232 converters (the little board says "USB2TT_CO4 from homsun-ic.com") which I got when ordering the SDS011. But of course, I have to make a cable for it first. Will take a couple of days (I don't have soldering equipment at home). And I also have no idea how to control the Tera NextPM using terminal 😅 
Guess I have to download that Windows Testing software from the official Tera website, but I think it's a little bit annoying that you have to provide an e-mail for every download...

What I could try in the meantime is compile and test the beta-piera branch... but I guess the code in there is the same, regarding serial communication?

pjgueno, 2022-02-22:
I have just asked Tera about those extrema. Ihave the same just for PM 10. Now the firmware check the checksum before using the data. It will prevent this if it is a problem with the serial.

@str3yt @sumerland 
Check the new full implementation of the NextPM in the beta of the firmware. I added a fulltime and a screen to check state, T° and RH in the sensor.

sumerland, 2022-02-22:
Brilliant - compiling now, will flash tomorrow.

pjgueno, 2022-02-22:
No spikes no more !
[image]
Looks good. I bet it was a serial problem on the last bytes.

pjgueno, 2022-02-22:
Select Debug = 5 because the ESP8266 seems to be very sensitive for 115200 8E1 which the Next PM receive and send.

pjgueno, 2022-02-23:
Still a mistake I just corrected:

Line 167 in airrohr-firmware.ino => bool npm_fulltime = NPM_FULLTIME;

sumerland, 2022-02-24:
I added the fix above yesterday and flashed a moment ago. I assume the option "Next PM fulltime" is meant as an optional switch, i.e., enable both "Tera Sensor Next PM (particulate matter)" AND "Next PM fulltime"  to have the heating on all the time.

Looking forward to saying good-bye to the spikes :-)

pjgueno, 2022-02-24:
Yes. The both

sumerland, 2022-02-26:
Running in fulltime mode since yesterday. Not sure it improves the result. There seems to be another problem leading to a temporary shift in overall PM2.5 level. NPM in red:

[image]

PM10 is at least in line with the Sensirion SCS (blue)
[image]

pjgueno, 2022-02-26:
> PM10 is at least in line with the Sensirion SCS (blue)

it is strange because both are exterpolations from the PM2.5 in the SPS and the NPM...

>There seems to be another problem leading to a temporary shift in overall PM2.5 level

I have triple checked the "word" expression it is actually the right bytes and the fulltime is just a deactivation of the switch function for the "stop after measure". What does the state code and the T/RH in the NPM say ?

sumerland, 2022-02-26:
[image]

What puzzles me is that the NPM does not detect jumps in PM like the rest, neither in PM2.5 nor in PM10

[image]

str3yt, 2022-02-28:
Havent updated the firmware since I commented here last time, so you can still see a hiccup in PM10.
Three different sensors have been running in my room now for a while. Ordered an ESP32 evalboard by the way to try a ucontroller with better serial interface...
[image]
[image]