opendata-stuttgart/sensors-software #795: Internal Server Error for sensor upload

chansters, 2020-09-16:
My sensor hasn't been able to upload data for past 10 hours or so (since just before 1pm Pacific Standard Time).  https://devices.sensor.community/sensors is showing Internal Server Error.  My sensor # is 49528.

ricki-z, 2020-09-16:
The main database server is down because of a hardware failure. The restore of nearly 1TByte needs some time.

chansters, 2020-09-16:
> The main database server is down because of a hardware failure. The restore of nearly 1TByte needs some time.

Thank you !

sumerland, 2020-09-16:
Does the server issue also explain the gaps in the grafana plots? Opensensemap uploads are not affected.
[image]

sumerland, 2020-09-17:
Plots are looking good again. Many thanks for fixing this!

closed 2020-09-17 by ricki-z