opendata-stuttgart/sensors-software #802: Can't see my data from sensor

kraptak, 2020-10-07:
I have a SDS011 and DHT22 sensor.  All worked fine till the end of september. I can connect by wifi to the sensor. 
Readings are ok but nothing shows on madavi.de or other luftdaten apps.

In section 'errors only' I have:
WiFi  0
OTA Return  200
Data Send  48
Data Send Return  307
SDS011  0

Firmware is:
NRZ-2020-129
ST:1
SDK:2.2.2-dev(38a443e)
Core:2.6.2=20602000
lwIP:STABLE-2_1_2_RELEASE
glue:1.2-16-ge23a07e
BearSSL:89454af 

Please help

dirkmueller, 2020-10-07:
@kraptak any idea what you changed end of september? 

can you try "use beta firmware" flag and report if that fixes anything? if not, please post screenshots of the "device status" page and optionally the /debug page (after it slowed down updating)

ricki-z, 2020-10-07:
And could you please supply the chipID? That gives me a chance to check some of our logs.

kraptak, 2020-10-08:
> @kraptak any idea what you changed end of september?
> 
> can you try "use beta firmware" flag and report if that fixes anything? if not, please post screenshots of the "device status" page and optionally the /debug page (after it slowed down updating)

Than you for your advice. I enabled "use beta firmware" yesterday. Some data showed but only for about 2 hours.
My sensor is esp8266-3986292
Here is some data: 
https://api-rrd.madavi.de/grafana/d/GUaL5aZMz/pm-sensors?orgId=1&var-chipID=esp8266-3986292&theme=light

I noticed that the time of my sensor is 2 hour earlier that real is. I live in Poland  - sensor shows 19:04 and in real it is 21:04.
Maybe that is the problem?
Below is sensors actual state.
[image]

kraptak, 2020-10-08:
> And could you please supply the chipID? That gives me a chance to check some of our logs.

:) Yes of course: 
esp8266-3986292

dirkmueller, 2020-10-09:
@kraptak so we can see that you have 307 out of 562 sensor data send failures, and the last one was ending in http return code 307, which is a http redirect. is it possible that your wifi router/access point is treating the client not as allowed to be connected to the internet freely?

kraptak, 2020-10-10:
> @kraptak so we can see that you have 307 out of 562 sensor data send failures, and the last one was ending in http return code 307, which is a http redirect. is it possible that your wifi router/access point is treating the client not as allowed to be connected to the internet freely?

Thank you Dirk! It was a very accurate diagnose :) Something happend to my WAN LTE modem and I have to upgrade it with a new firmware. Maybe the configuration was affected by the storm. All was looking good till and I restarted it several times. Upgrade helped.
Is there a list of errors of my sensor somewhere? Could you give me a link?
Thanks again. Have a nice day!

dirkmueller, 2020-10-10:
> Is there a list of errors of my sensor somewhere?

well, its the screenshot you posted above. those errors are not sent anywhere, that would not be very privacy friendly ;)

thanks for reporting back!

closed 2020-10-10 by dirkmueller