opendata-stuttgart/sensors-software #804: Looking for Ideas on How to Increase Wifi Reception on the ESP8266

DCBene, 2020-10-10:
Hello again,
I'm coming to you today with an issue that some of you have also encountered - the very limited reception range of the onboard antenna of the ESP8266. I've been looking into ways on how to increase the Wifi-reception of the node for quite a while now. The only sollution other folks (outside of this forum) operating the NodeMCU V3 ESP8266 seem to have found, is soldering an antenna-cable to the platform itself. One thing that bugs me about this, are my own rather terrible soldering capabilies. On top, I was simply hoping for a more elegant sollution to add an external antenna.
If any one of you has had the same and found a better sollution, I'm very much looking forward to hearing from you.
Thank you and Cheers!
Benedikt

ricki-z, 2020-10-11:
Hello,
there is the WeMos D1 mini Pro (i.e. here: https://nettigo.eu/products/wemos-d1-mini-pro-wifi-module).
This should be working with our firmware, but we haven't tested this. You need to check that the module is configured for the external antenna.

DCBene, 2020-10-23:
Hey Ricki,
Thank you for the advice. Much appreciated. Will look into it and leave feedback.
Cheers!

DCBene, 2020-10-23:
Am also looking into trying these on the NodeMCU ESP8266: https://www.ebay.de/itm/2-x-NRF24L01-PA-LNA-SMA-Antenne-Transceiver-Modul-2-4G-Antenna-Long-Range/252713959128?ssPageName=STRK%3AMEBIDX%3AIT&_trksid=p2057872.m2749.l2649

Just a bit worried about tweeking you original firmware to get it running. Haven't done any real coding in my life so far...

RikDrabs, 2020-10-23:
Looks this NRF24L01 is a stand-alone module for communications - not WiFi - over longer distances. And you need 2 modules to establish a communications channel, one at each side of the link, and connected to an Arduino or 8266 controller over SPI, at each side. This is - again - not WiFi. The frequency used is the same, but this is the only common ground.
The library for using this on Arduino/ESP8266? is called RF24.
See: https://lastminuteengineers.com/nrf24l01-arduino-wireless-communication/
See also: http://acoptex.com/wp/basic-project-083b-esp8266-esp-12e-module-and-nrf24l01-2-4ghz-rf-transceiver-module/
as an example of what can be achieved with NRF24L01 module, and that is not WiFi extension, but a point to point link over 2.4 GHz.

I'm sure Ricki's solution works. It's a bit more work at first glance, but I think it's the shortest workable route to range extension for the Luftdaten sensor. Maybe a PCB, to connect the parts reliably, would make things easier...

@Ricki, thanks for the advice!

DCBene, 2020-10-24:
Hey there @RikDrabs - thank you for the insight! I greatly appreciate it. Will look into @ricki-z's suggestion for sure and update you on the progress. Thanks again!

WalterPepeka, 2020-11-01:
You may also use the LoRa technology, but you need extra modules.
LoRa is perfect for IOT, low consumption, veeeeery long range
Look at this excellent youtube channel :
https://www.youtube.com/playlist?list=PL3XBzmAj53Rkkogh-lti58h_GkhzU1n7U

ricki-z, 2020-11-01:
LoRa isn't available everywhere (without building an own gateway), while most users have a wifi network.
That's why our priority is on wifi.

buckfast-beekeeper, 2020-11-02:
> 
> 
> Hello,
> there is the WeMos D1 mini Pro (i.e. here: https://nettigo.eu/products/wemos-d1-mini-pro-wifi-module).
> This should be working with our firmware, but we haven't tested this. You need to check that the module is configured for the external antenna.

Wemos D1 mini pro works like a sharm. I use it since 28/3/2020

Lolin Wemos D1 mini pro v2 is the same newer board with a battery charger on board.  

[https://www.aliexpress.com/item/32909702224.html](url) replace the 0 Ohm resistor from R16 to R15 to use the external antenna.
[sch_d1_mini_pro_v1.1.0.pdf](https://github.com/opendata-stuttgart/sensors-software/files/5474411/sch_d1_mini_pro_v1.1.0.pdf)

[https://www.aliexpress.com/item/32724692514.html](url) idem for the antenna resistor.
[sch_d1_mini_pro_v2.0.0.pdf](https://github.com/opendata-stuttgart/sensors-software/files/5474437/sch_d1_mini_pro_v2.0.0.pdf)

glosair, 2020-11-24:
You can use the ESP-07 (but it has a smaller flash size you can upgrade this if you are good at soldering) and attach it to one of the many custom boards that are available like this one https://oshwlab.com/t.ballam/partical-sensor-v1-3