opendata-stuttgart/sensors-software #808: ADD More sensors BME680 and HM3301

julienrat, 2020-10-13:
Hi !
Is it possible to add BME680 and HM3301 in available sensors ?
And if you don't have the time, how to add new sensors ?
Thx in advance !

https://github.com/Seeed-Studio/Seeed_PM2_5_sensor_HM3301

https://github.com/adafruit/Adafruit_BME680

dirkmueller, 2020-10-14:
see #411 for BME680. I started to look at it (I have a sensor myself), but the air quality sensing part is unfortunately closed source and I wasn't able to find reliable information on how to reimplement that in open source without violating some trademark somewhere. I think the sensor is far too expensive to make it just a drop in for a BME280, without gas it makes no sense. 

HM3301 sounds interesting, any idea how reliable the data quality is ? It might be a similar design to the sensors we support already (which means it is susceptible to the same flaws potentially)

julienrat, 2020-10-14:
Hi !
I have made a quick and dirty lib for HM3301 sensor !

[
[H3301_dust_i2c.zip](https://github.com/opendata-stuttgart/sensors-software/files/5378819/H3301_dust_i2c.zip)
](url)