opendata-stuttgart/sensors-software #810: No JSON data after 17 Oct update?

heinemancp, 2020-10-18:
This weekend I get error after error about domoticz can't open /data.json at the sensor. I tried it in a browser, indeed. No connection. Other connections (/values, /config) work. Is the JSON data broken? 
Plz redo, I used this to log my own data at my domoticz instance. Thnx

ricki-z, 2020-10-18:
I have checked it here. It's working. Can anyone else confirm this problem?
Can you send us the list of activated sensors (may be caused by a sensor type we haven't tested)

heinemancp, 2020-10-18:
{"software_version": "NRZ-2020-131", "age":"72", "sensordatavalues":[{"value_type":"SDS_P1","value":"0.30"},{"value_type":"SDS_P2","value":"0.00"},{"value_type":"temperature","value":"13.40"},{"value_type":"humidity","value":"99.90"},{"value_type":"BMP280_pressure","value":"102437.75"},{"value_type":"BMP280_temperature","value":"14.85"},{"value_type":"samples","value":"7438064"},{"value_type":"min_micro","value":"33"},{"value_type":"max_micro","value":"20126"},{"value_type":"interval","value":"300000"},{"value_type":"signal","value":"-86"}]}

Allright, just tested it again and again. Sometimes it works, mostly not. Up this weekend no problem. Connection is ok. There is some problemen opening the JSON page.

heinemancp, 2020-10-18:
In my opinion it looks like the device is to busy. just reseted the sensor. 

Status:
 
 Apparaatstatus
Parameter	Waarde
Firmware-versie	NRZ-2020-131
ST:1
SDK:2.2.2-dev(38a443e)
Core:2.7.3-3-g2843a5ac=20703003
lwIP:STABLE-2_1_2_RELEASE
glue:1.2-30-g92add50
BearSSL:5c771be 
Free Memory	31720 
Heap Fragmentation	4 %
Last OTA	72s 
NTP Sync	1 
NTP Info	194.5.96.11 (0.pool.ntp.org) reachable: Yes 
Tijd (UTC)	Sun Oct 18 12:18:41 2020  
Uptime	72s 
Reset Reason	Software/System restart 
 
SDS011	18-11-16(40d8) 
 
enkel foutmeldingen
WiFi	0 
OTA Return	200 
SDS011	0

heinemancp, 2020-10-18:
I set the debug level from minimal (is that standard in the new UI?) to none. Seems to help a bit. Reinforced my idea for a too busy sensor since the update.

holgerbohni, 2020-10-19:
As I wrote in Issue 814: 
Only every third measurment is done. 
web interface unreachable most of the time
Lots of errors.
Is there a way to switch back to firmware from january?

dirkmueller, 2020-10-19:
@heinemancp can you test if disabling BMP280 in your /config fixes the availability issue of your /data.json?

what is visible here is a very low wifi signal (-86). is there a way you can retest this temporarily nearer the wifi access point?

heinemancp, 2020-10-20:
Just disabled the BMP280 in my config. Will see, also turned on the logging of the wifi data. the connection will remain up, both the connection will be dropped. i see this also in my ping-data. but will give some more insights at the end of this day. 

2020-10-20 08:39:54 | On
2020-10-20 08:38:54 | Off
2020-10-20 08:37:49 | On
2020-10-20 08:36:48 | Off
2020-10-20 08:32:43 | On
2020-10-20 08:31:42 | Off
2020-10-20 08:19:33 | On
2020-10-20 08:18:32 | Off
2020-10-20 07:55:47 | On
2020-10-20 07:53:34 | Off
2020-10-20 07:52:25 | On
2020-10-20 07:51:21 | Off
2020-10-20 07:36:11 | On
2020-10-20 07:35:04 | Off
2020-10-20 07:31:46 | On
2020-10-20 07:30:40 | Off
2020-10-20 07:11:09 | On
2020-10-20 07:10:03 | Off
2020-10-20 07:02:24 | On
2020-10-20 07:00:10 | Off
2020-10-20 06:57:54 | On
2020-10-20 06:56:47 | Off
2020-10-20 06:51:21 | On
2020-10-20 06:50:16 | Off
2020-10-20 06:43:42 | On

heinemancp, 2020-10-20:
Ping stats: 
2020-10-20 10:35:49 | Off
2020-10-20 09:42:52 | On

with error opening file: 
2020-10-20 10:15:10.386 Error: Error opening url: http://[IP]/data.json

image of QoS:
[https://photos.app.goo.gl/kdfUTqD5EGkgowL17](https://photos.app.goo.gl/kdfUTqD5EGkgowL17 )

heinemancp, 2020-10-20:
I reenabled the BMP280 sensor, the connection is dropping more, allthough the ping stats are OK:

2020-10-20 11:05:11.076 Error: Error opening url: http://[IP]/data.json
2020-10-20 11:16:30.475 Error: Error opening url: http://[IP]/data.json
2020-10-20 11:20:03.478 Error: Error opening url: http://[IP]/data.json

Date | Data
2020-10-20 10:55:14 | On
2020-10-20 10:54:13 | Off
2020-10-20 10:52:08 | On
2020-10-20 10:51:08 | Off
2020-10-20 10:48:03 | On
2020-10-20 10:47:02 | Off
2020-10-20 10:43:57 | On
2020-10-20 10:42:57 | Off
2020-10-20 10:36:49 | On

dirkmueller, 2020-10-21:
This sounds like #814  please look there, looking for feedback.

heinemancp, 2020-10-22:
OK, will watch that thread. Can close this one.

closed 2020-10-22 by heinemancp