opendata-stuttgart/sensors-software #811: Possible to remove unnecessary DEBUG Info?

imaluschi, 2020-10-18:
Possible to reduce DEBUG-Info when choosing max. Info?
Would give better overview when investigating problems.

[image]

At least, i don't understand why it is useful to put out the same message every second.
As far as i know SDS is not running during some seconds before the measurements (warming up).

[image]

It seems every second the DEBUG-Infos is written.

[image]

Maybe add a DEBUG-message when the SDS is beginning to warm up? And also something like "SDS011-Measurements" when beginning to measure?

ricki-z, 2020-10-18:
In case of a software crash we need to know what the firmware has done at this moment. Debugging ESP866 firmware is hard enough even with that info.

imaluschi, 2020-10-18:
I belive it's hard enough.
The problem for me as user is to get informations why something is not working. Would also be helpful for you to get such informations in case of logging an Issue.

For example, i updated my Sensor (SDS011, DHT, BME280, DNMS) from a beta version with the latest release (NRZ-2020-131). Some time later, i saw, that the DHT was not sending data anymore.
Before i file an Issue, i want to have more informations (as you too ;-) ). 

I didn't even touch a cable...

[image]

[image]

Similar on another Device with only DNMS and short time Intervall with sending to an InfluxDB. But much better overview without SDS-messages.

[image]

[image]

I don't know what is possible, but you can imagine that the informations are not very helpful to find a cause.

I will have an eye on these problems and file an Issue in case it occurs more often.

dirkmueller, 2020-10-19:
@imaluschi I agree that "max info" is usually printing too much info. however, for debugging why there are failures, I think you don't need to set "max info", the default min-info should cover those. 

In cour case we can see the key issue being

DHT11/DHT22 read failed

this is when the sensor doesn't return valid data. can you try powering off this node for like a minute and then powering it on again? does this recover it?

if it doesn't, we might have an issue with the change of default I2C speed that was introduced in 2020-131. which firmware version was the last one that previously worked?

imaluschi, 2020-10-19:
Ok, thanks, i will use min-info in future.
I filed a new issue concerning the DHT failure - #816 .

For my case, we can close this issue.

closed 2020-10-19 by imaluschi