opendata-stuttgart/sensors-software #815: BME280 pressure - wrong decimal places in data.json

ottk3, 2020-10-19:
It seems that the decimal places in the data.json are off by to places for the pressure of the BME280.
While the pressure value of **1009.37** hPa in the dashboard looks valid, the data.json value is **100936.53** hPa.

**dashboard:**
[image]

**data.json:**

	{
	  "software_version":"NRZ-2020-131",
	  "age":"24",
	  "sensordatavalues":[
		{
		  "value_type":"BME280_temperature",
		  "value":"7.98"
		},
		{
		  "value_type":"BME280_pressure",
		  "value":"100936.53"
		},
		{
		  "value_type":"BME280_humidity",
		  "value":"89.82"
		}
	  ]
	}

ricki-z, 2020-10-19:
And again, we send and save the pressure data in Pascal. This is the unit we get from the sensors and we try to save all data as raw values.

ottk3, 2020-10-19:
Oh! I wasn't aware that the raw value is in Pascal - my bad.

closed 2020-10-19 by ottk3