opendata-stuttgart/sensors-software #816: DHT sometimes fails since update to NRZ-2020-131

imaluschi, 2020-10-19:
Since the last update the DHT fails sometimes.

[image]

Today (19th of oct.) it was not possible to get the DHT working again with a  sensor restart. Tried it two times with no success. Had to unplug the USB-cable from the power source and after starting the DHT worked again.
[image]

Today evening i realized that again DHT is not working.
[image]

This time it was possible to get it working again with a sensor restart.

Device status from today
[image]

Really don't know what the problem is, no hint by debug-messages, only that the sensor could not be read. The problem occured also yesterday.
But it seems that the problem occured already with NRZ-2020-130-B10/B11 (not sure what version i had on 15.10. and when i updated to B11).

imaluschi, 2020-10-19:
It's hard to say, with which version the sensor-error occured first, when i read an comment of #789 it seems, that already with NRZ-2020-130B9 have been a problem with DHT.
https://github.com/opendata-stuttgart/sensors-software/issues/789#issuecomment-703054268

dirkmueller, 2020-10-19:
@imaluschi this issue report shows 0 wifi disconnects over 12 hours, so it is very likely a different problem.

imaluschi, 2020-10-21:
It seems to be a specific problem with the DHT sensor.
The last two days no action was needed/done, the DHT failed but began to work again.
The high amount of errors sending to sensor.community comes probably from using more sensors than i could register (SDS011 and DNMS are registered, DHT and BME280 not).

[image]
[image]
[image]
[image]

And sometimes data of both sensors are not sent (or failure of both?? I don't know), but data were sent because data of DNMS is available during the time of missing DHT/BME values. Same problem or another problem??? Device status as above for that time.
[image]
[image]

imaluschi, 2020-10-21:
> @imaluschi this issue report shows 0 wifi disconnects over 12 hours, so it is very likely a different problem.
@dirkmueller, yes i think so too. Some of the problems disappeared since NRZ-2020-130B10 but some remained.
I mentioned my post in the WLAN-Issue here because it seems the DHT-problem occured already (at least) with B9.

[image]
As you can see, missing data of both sensors was the main problem at this time. Sometimes "only" DHT and BME where missing, somtimes all values (DHT, BME, SDS and DNMS).
DHT errors where minor but present.
[image]
[image]

dirkmueller, 2020-10-26:
@imaluschi could you please retest with the current OTA in beta channel? Would be good to know if this issue happens to be fixed by that.

imaluschi, 2020-10-27:
Done. Firmware: NRZ-2020-132-B2/DE (Oct 23 2020)

imaluschi, 2020-11-03:
Beta-Firmware NRZ-2020-132-B2/DE is not perfect (concerning this issue) but much better.
Some short outages.

[image]
[image]
[image]

But yesterday again both BME and DNMS stopped working, UI was not very responsive (only sometimes, i think around measurements). More informations in #739 .

imaluschi, 2021-01-18:
I didn't realize missing DHT values for quite weeks.
Firmware: NRZ-2020-133/DE (Nov 29 2020)
In my opinion we could close this issue.

closed 2021-01-18 by imaluschi