opendata-stuttgart/sensors-software #817: Hotspot mode timeout

iceburnenator, 2020-10-21:
I have the following situation:
After a power outage my router boots slower than the sensor.
Unable to find a suitable network the sensor enters hotspot mode and remains in that state forever.
I need to power cycle the sensor in order to recover its operation.

I would suggest adding some timeout that will reboot the sensor after about 5-10 minutes while in hotspot mode.

dirkmueller, 2020-10-22:
It's actually supposed to be doing that (and it worked last time I tried, which was a few weeks ago). the behavior it does is: 

* try up to 40 seconds to connect to the preconfigured wifi
* when fails, enter hotspot (access point) mode
* for configurable router interval (by default 10 minutes) it remains in that mode
* afterwards it will go back to endless retrying the configured wifi

this cycle only starts again on a manual power cycle or after 28 days of uptime.

dirkmueller, 2020-10-22:
Just retested, it works as described.

iceburnenator, 2020-10-22:
Thank you for the quick response.
That is quite strange.
The problem started back in February when the sensor stopped reporting data to the server. At first I thought that the ESP board is dead because a simple power cycle did not solve the problem. But could not test it further as it mounted in a hard to reach location. I've actually ordered new boards in order to directly replace it once I had a chance. Strangely enough it came to live all by itself after a power outage about a month later.
The same happened for a second time back in June. And then again last month. A few days later I've spend about half an hour power cycling the sensor and the router and that is how I noticed hat I reported earlier.
Will wait for this to happen again and this time I will look for the hotspot SSID and check router Wi-fi registration attempts for at least 30 minutes in order to gather more info.

dirkmueller, 2020-10-22:
so it is possible that there is a bug, especially if you changed the router mode timeout to a very large value (I can see that we don't handle overflow correctly). 

also, I mean it is theoretically thinkable maybe that there are other devices that keep accessing the open hotspot, which will extend the timeout. 

do you know how the timings are roughly, e.g. how long does it take your router to reestablish wifi after power loss? maybe I can hit this situation then here with reproducing. 

to avoid interest collissions with my family I can only do those wifi tests when noone else is using it here during the pandemic times ;)