opendata-stuttgart/sensors-software #820: I can't set the time.The time on the sensors is 3 hours less than in my time zone.Tell me this time where is the server located or my time? And how do I change it if it's my time?

Anton8998, 2020-10-22:
[image]

holgerbohni, 2020-10-22:
Hi @Anton8998, as far as I know, you can't change it. All devices are synchronized with UTC, that means your display/app/database has to manage your timezone. If you try to perform some analysis on the data, it's quite useful and simple to have all recorded values in UTC to prevent timezone confusions later on.

dirkmueller, 2020-10-22:
@Anton8998 by intention we keep all the sensor.community devices synced in time with UTC. it helps with troubleshooting and with data correlation globally. we do need to keep some time on the devices as they need to validate TLS certificates, so correct system time is essence. 

it should not cause any side effects for anything anywhere however.

closed 2020-10-22 by dirkmueller