opendata-stuttgart/sensors-software #821: Issue with measured values(temp and humidity) from BME280 and DHT22, why this happens

krasigarchev, 2020-10-25:
Dear team,

I'm realy new in github and IoT, please help me whith this issue. 
DHT22 and BME280 has diference in measured values.
You can found me here or in telegram @Krasi_Garchev

Sensor	Parameter	Value
SDS011	PM2.5	16.3 μg / m2
SDS011	PM10	24.2 μg / m2
 
DHT22	Temperature	12.7 ° C
DHT22	Air humidity	99.9%
 
BME280	Temperature	10.9 ° C
BME280	Atmospheric pressure	1004.23 hPa
BME280	Air humidity	84.7%
 
WiFi	Signal strength	-49 dBm
WiFi	Signal quality	100%

pjgueno, 2020-10-27:
The DHT22 almost always blocks at 99,9% after a while. It is the reason why we advise people to use the BME280 which works a lot better.

ricki-z, 2020-10-27:
Another factor is the location of the sensors inside the case. The electronic part are creating some heat so that the upper part of the case is normally warmer than the lower part.

linuxlurak, 2021-01-04:
Would you actually place the BME280 inside the sensor? Or are you talking specifically of the airrohr version with a HT Bogen? I placed it normally outside my own boxes (no HT-Bogen) with some kind of weather protection. 

On a hackerspace website I saw they placed the BME280 inside a case. Not sure why as it only measures the enclosed environment of the sensor inside the box.

imaluschi, 2021-01-18:
If you place the temperature sensor (DHT or BMP/BME) inside the case (box), you have at least 3°C more due to power consumption of the ESP. Can't say how the measurements will change in a box only for BME.

Concerning the issue with wrong values of BME you can find something here: #739 
In my opinion it's a problem with I2C. I built 4 airrohr-sensors with SDS011, DNMS, BME280. In two cases (one is my own) DHT in addition. The problem occured in all 4 airrohr-sensors, even tough i replaced the BME in two different sensors. I also can't detect a trigger. In most cases, the airrohr firmware get stuck after some time.
For the last airrohr-sensor I added a DHT to easily remove BME if the problem (with firmware-crashes) occurs "too often" and still have temp./humidity (even tough it's less precise).

At one of the sensors the BME values are often wrong, but at least the software don't get stuck, so the (important) measurements of PM and noise are still running.

[image]

umweltsensor, 2021-01-18:
Similar situation like @imaluschi's in my case. Have 3 Sensors, two of whitch are SDS011,BME280,DNMS one without DNMS.
One Sensor runs very stable (esp8266-586719).

Following sensor (esp8266-7876127) goes south at some point. I can't identify the problem only some symptoms:
[image]

BME280
07.01.2021 14:35:00 T:6  H:52% P:950
07.01.2021 14:45:00 T:-145 H:100% P:548

DNMS
07.01.2021 14:29:00 LAeq: 59.23
No next measure point, should have been at 14:39:00

BME280 stopps (Temp: -145°C; Humidity 100%; Pressure 548 HPa). DNMS stopped working too.