opendata-stuttgart/sensors-software #822: Add Support For A Geiger Counter

thetruebumblebee, 2020-10-27:
Hi Folks,

thanks for your great work, I like it very much!
How about adding a geiger counter as described in
https://www.heise.de/developer/artikel/Strahlende-Aussichten-3317716.html 
(german). Since I am not an experienced programmer, how can I help?

Kind Regards
Tom

pjgueno, 2020-10-27:
The data of the multigeiger project should be already collected in our archive :
https://ecocurious.de/projekte/multigeiger-2/
https://github.com/ecocurious2/MultiGeiger
https://multigeiger.citysensor.de/

thetruebumblebee, 2020-10-27:
Hello Pierre-Jean,
thanks a lot. I was just hoping to get it all together in one device. Just an additional sensor to configure in the best case. Or did I miss st?
Kind Regards
Tom

pjgueno, 2020-10-27:
Some people are currently trying to do a stable ESP32 firmware for Sensor.Community. In this case it shoud be possible to put everything on the same board because the Multigeiger use an ESP32 with or without Lora (the brand is Heltec). But I am not sure that you can configure enough UART pins for all the sensors. And the firmware shall be too big to allow the auto-update. Maybe It could be easier to program the Multigeiger on an ESP8266...
Best regards