opendata-stuttgart/sensors-software #825: Sensor not seen in map https://maps.sensor.community/?nowind#16/48.7996/9.8027

maselmann, 2020-10-31:
The sensor ist registered under https://devices.sensor.community/sensors/register, but either not listed there nor to be seen in the map under the given location ("Your location : LatLng(48.79594, 9.800202").
Date is available in https://api-rrd.madavi.de/grafana/d/GUaL5aZMz/pm-sensors?orgId=1&theme=light&var-chipID=esp8266-14019113
What can be done for showing the sensor in the map? And in my list of registered sensors? 
A few days ago both the map and the list were OK.

Regards Michael

pjgueno, 2020-10-31:
Hi. Please send me the lsite of the registered sensor of your account under this link:
https://devices.sensor.community/sensors?lang=en
It should looks like the attachment.
[image]
Then check the geocoodinates under "Setting". A lot of sensors are at 0:0 in the Gulf of Guniea...
If there is no sensor in the list, you have not registered anything yet. Did you really fill all the fields and save at the end.
**Send me a direct message**. I have another solution.

ricki-z, 2020-10-31:
@maselmann the sensor should be available again at devices.sensor.community and on the map.
As you may remember there was some confusion about the registration of your sensor (wrong account, wrong chipID,...).
@pjgueno this is solved.

closed 2020-10-31 by ricki-z