opendata-stuttgart/sensors-software #828: Sensor refuses to connect to WiFi

maltehuebner, 2020-11-09:
Hej,

I’m not sure if this is the right place for this kind of question. I bought two of these kits: https://nettigo.eu/products/sensor-community-kit-sds011-bme280-english-language

I assembled one of them and everything worked fine. I was able to connect to the sensor via wifi, did the setup, and could connect it to my wifi at home. I registered the sensor at https://devices.sensor.community/ and it’s still listed there as it connected at 2020-11-08 16:21:12 UTC. Unfortunately, it disconnects right afterward and restarts in access point mode. When I connect to the sensor via browser and submitted the right credentials to log into my wifi, it tries to connect to wifi, fails, and opens the access point mode again.

This is what I got from the debug output:

```
��怘����f��`f��~`f���怘���fx�fx����xf�~x���f��)�
                                               �@H��airRohr: NRZ-2020-131/EN
mounting FS...
opened config file...
parsed json...
output debug text to displays...
Saving config...
Config written successfully.
output debug text to displays...
Restart.
���ԅ��D�BairRohr: NRZ-2020-131/EN
mounting FS...
opened config file...
parsed json...
output debug text to displays...
Connecting to Caldera
........................................
output debug text to displays...
Starting WiFiManager
AP ID: Feinstaubsensor-2958133
Password:
scan for wifi networks...
AP Password is:
Starting Webserver... (IP unset)
ws: not found ...
ws: not found ...
ws: not found ...
ws: config page ...
begin webserver_config_body_get ...
wifi networks found: 16
ws: wifi ...
ws: config page ...
output debug text to displays...
Saving config...
Config written successfully.
output debug text to displays...
Restart.
H!�������airRohr: NRZ-2020-131/EN
mounting FS...
opened config file...
parsed json...
output debug text to displays...
Connecting to Caldera
........................................
output debug text to displays...
Starting WiFiManager
AP ID: Feinstaubsensor-2958133
Password:
scan for wifi networks...
AP Password is:
Starting Webserver... (IP unset)
ws: not found ...
ws: not found ...
ws: not found ...
ws: not found ...
ws: not found ...
ws: config page ...
�怘���fx�fx����xf�~x���f���.........................
output debug text to displays...
Starting WiFiManager
AP ID: Feinstaubsensor-2958133
Password:
scan for wifi networks...
AP Password is:
Starting Webserver... (IP unset)
ws: not found ...
ws: not found ...
ws: config page ...
begin webserver_config_body_get ...
wifi networks found: 10
ws: wifi ...
ws: config page ...
output debug text to displays...
Saving config...
Config written successfully.
output debug text to displays...
Restart.
��)�
   �@H�airRohr: NRZ-2020-131/EN
mounting FS...
opened config file...
parsed json...
output debug text to displays...
Connecting to Caldera
........................................
output debug text to displays...
Starting WiFiManager
AP ID: Feinstaubsensor-2958133
Password:
scan for wifi networks...
AP Password is:
Starting Webserver... (IP unset)
ws: not found ...
ws: not found ...
ws: not found ...
ws: config page ...
begin webserver_config_body_get ...
wifi networks found: 9
ws: wifi ...
ws: config page ...
output debug text to displays...
Saving config...
Config written successfully.
output debug text to displays...
Restart.
)�
 �@H�airRohr: NRZ-2020-131/EN
mounting FS...
opened config file...
parsed json...
output debug text to displays...
Connecting to Caldera
........................................
output debug text to displays...
Starting WiFiManager
AP ID: Feinstaubsensor-2958133
Password:
scan for wifi networks...
AP Password is:
Starting Webserver... (IP unset)
```

Does anyone have an idea of how to connect the sensor to my wifi again?

Thanks,
MAlte

pjgueno, 2020-11-09:
Hi,
Caldera is the name of your Wifi-Network, isn't it?
The sensor can't connect to it. Did you double-check the password. Are you sure your really succeed to be conected and to see data at devices.sensor.community ? Did someone change the Wifi-password? Are there some very special characters in it? How far is the sensor from the access point?

Here I can see the sensor :
https://api-rrd.madavi.de/grafana/d/GUaL5aZMz/pm-sensors?orgId=1&theme=light&var-chipID=esp8266-2958133

It stopped on the 8th at 23.30 an started again for a few minutes on the 9th at 00.30.

But I would also need the Sensor ID on the map to check another thing.

In my opinion, the router is to far away from the sensor.

pjgueno, 2020-11-09:
Another thing : is the WPS option on?

maltehuebner, 2020-11-09:
Thanks for your answer!

While you were typing, I assembled the second sensor, ran thru the setup process again, but did not allow it to self-update to the newest firmware version. It still runs on NRZ-2018-123B and looks fine: 

You’re right, Caldera is the name of my wifi. I use an AVM FRITZ!Repeater as an access point for the sensor and it’s just a meter and a wall between sensor and access point. The second sensor testifies a wifi signal strength of -54 dBm, a signal quality of 92 %. As I use a password manager to copypasta the wifi password, I’m totally sure it is correct.

The number of my first sensor is 51455, the second sensor registered as 51489. The first sensor sent at least one datum yesterday, which appeared in the luftdaten map as well as on my own website here: https://luft.jetzt/LFTDTN37526 The second sensor delivers it’s data here: https://luft.jetzt/LFTDTN37568

As the first sensor runs on the firmware released in october, while the second sensor did not autoupdate it’s firmware, maybe the whole thing is related to this issue? #814

MrBollie, 2020-11-12:
My Fritz!OS was updated to 07.21 last night and since then, the air sensor can't connect either. There are some wifi-related things in the changelog, like enabling WPA3 and mesh steering. Perhaps we're having a similar issue here.

maltehuebner, 2020-11-14:
I tried several networks like the one from my old FRITZ!Box 7362 with FRITZ!OS 07.12, the unsecured guest wifi, my neighbor’s wifi, and even my iPhone hotspot, the result is always the same: I set up wifi name and password in the sensor’s configuration screen, the sensor restarts, tries to connect, and enters it’s access point mode again.

As I layman, I really have no idea of nothing, but it would guess in my case the problem was the self-updating firmware of the sensor. Maybe I am able to flash the old firmware into the sensor again.

maltehuebner, 2020-11-27:
I really don’t understand: I connected the sensor to my notebook to flash an older firmware version into it and even before starting the flashing process it just connected to wifi and everything went fine. Whatever. Looks like this issue was kind of self-healing. I will try another wifi network tomorrow and if everything wents fine, I got two working sensors.

maltehuebner, 2020-11-28:
Okay, I got it: When I connect the sensor via USB to my notebook, it uses an internal network, not the wifi. So the problem remains unsolved :(

Phaze-III, 2020-11-29:
Could you try if an upgrade to the current beta helps? You can do that with an auto-update with the Load Beta option enabled when connected to your notebook.

ricki-z, 2020-11-29:
There is a new release version online (same code as latest beta NRZ-2020-132-B3)
@maltehuebner could you please try this version?