opendata-stuttgart/sensors-software #830: SDS11 Error

chemacuestajimenez, 2020-11-17:
[image]
Good night 
I just configured these sensors but the SDS11 gives a wrong reading. I live in a residential area and the contamination values should not be that high.
Anyone know what might be happening??
thanks

bertrik, 2020-11-18:
Probably not an error. The software apparently was able to read a value from the SDS011. The PM10>PM2.5 so that seems consistent at least. Why do you think this reading is wrong? Is is high but not impossibly high.

Perhaps you can swap the SDS011 for another one.
Is the fan running during measurement (once every 145 seconds)?

ricki-z, 2020-11-18:
You should also change the power supply as a trial. Some power supplies limit the available current. And this may lead to higher values.

chemacuestajimenez, 2020-11-20:
> Probably not an error. The software apparently was able to read a value from the SDS011. The PM10>PM2.5 so that seems consistent at least. Why do you think this reading is wrong? Is is high but not [...]
Hi 
I imagined that there would be less pollution since we are close to the beach and there is not much traffic in this area. The ventilation time is 145 seconds.
thanks for the contributions

chemacuestajimenez, 2020-11-20:
> You should also change the power supply as a trial. Some power supplies limit the available current. And this may lead to higher values.

I have tried several power supplies and a "power bank". 
Thanks for the contribution