opendata-stuttgart/sensors-software #831: wifi problems with NRZ-2020-132-B1/DE

taeubler, 2020-11-17:
after oct 17 wifi Bug, i still have Wifi Connection Problems with Firmware: NRZ-2020-132-B1/DE (Oct 20 2020). is there a known issue?

br,
markus

taeubler, 2020-11-22:
PS.: Could it be a problem with the temperature? because within my apartment i still have no problems. but when i put the sensor to my balcony - i will get the connection problems. (there was a longer distance within my apartment and the connection was OK)
Next Step: i tried it with different 3 Powersupply, because maybe a Powersupply creates the problem -> No... :-(
there is only a window (not even a wall between the router and sensor. 4 to 5m distance)

Phaze-III, 2020-11-23:
Did you also try the current beta NRZ-2020-132-B3/DE (Nov 13 2020) ?

taeubler, 2020-11-23:
oh, i didnt saw B3... - i tried that: NRZ-2020-132-B1-wifirevert_de.bin - at the moment it looks good with NRZ-2020-130-B10/DE. and also NRZ-2020-132-B3/DE looks very good :-) - is running since 2 hrs without problems.

Phaze-III, 2020-11-28:
If NRZ-2020-132-B3/DE is still running stable this most likely is the same issue as in #814.

taeubler, 2020-11-28:
works perfect with NRZ-2020-132-B3! looks like the wifi problems have been solved. Thanks a lot!

ricki-z, 2020-11-29:
There is a new release version online (same code as latest beta NRZ-2020-132-B3)

ricki-z, 2021-04-02:
@taeubler did you check the latest release? And can we close this issue?

closed 2021-04-07 by ricki-z