opendata-stuttgart/sensors-software #835: sds011 problem wrong messurement

dankyn, 2020-11-25:
Good morning my sds011 has a wrong messurement, with windows software the messurement are correct.In your opinion what could be the problem.
[image]

ricki-z, 2020-11-25:
If you used our schematics then you did connect the pins GND and 5V on the SDS011 to the two pins labeled 'RSV' in the lower left corner of the picture. But this way the SDS011 won't get the power needed. In this case use the pins 'Vin' and 'GND 'in the lower right corner.
(The pisture is showing version V2 of the NodeMCU module)

dankyn, 2020-11-25:
Perfect it worked, thank you. I can ask why using a 4 pin bmp280 sensor the humidity is not detected, it is not really present in the menu, thanks again.

ricki-z, 2020-11-25:
In this case you got a BMP280. The BMP280 only measures temperature and humidity, while the BME280 also measures air pressure.

dankyn, 2020-11-25:
Sorry I do not understand, I have a bme280 and measures temperature, humidty and air pressure.
but only the temperature appears in the software selecting correct sensor.

ricki-z, 2020-11-25:
You only wrote about missing humidity (which is exactly difference between a BMP280 and a BME280).
Could you please post here a screenshot of the 'actual values' page, the config page of your sensor and a picture of the BME280?